nkolban/esp32-snippets #1017: BLE Server Get RSSI, MAC, Name etc

TheUruguayo, 2020-10-09:
Hello everyone, 
I want to do something that  seems  really elemental to a server, but at this point I'm not available to do.
**I want my BLE server to report, the RSSI, MAC and name of connected devices** (first 2 most of all). The scan is not good for me, I need to know which ones are connected.
It seems that the function get RSSI is implemented in the client side (which I don't need) is there a way of doing this. I've seen a post that does the following (https://github.com/nkolban/esp32-snippets/issues/893), but is not working for me, the values of that gives are apparently random, (RSSI and MAC, are very different from the scan).
I want to call a function, lets say _myserver.getRssi()_ or simply _getRssi()_, and have the RSSI and mac of a device connected to me. At this stage only one client is connected.  I do not understand how this libraries work yet, and how to move functions around, so any tips on that regard will be appreciated. 
(I'm working on a robot that follows this signal).

#include <BLEUtils.h>
#include <BLE2902.h>
#include <BLEScan.h>

String valor;
int aleatorio1;
int aleatorio2;
String alea1;
String alea2;
String alea3;

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

//para el servicio 2
#define SERVICE2_UUID        "45ecddcf-c316-488d-8558-3222e5cb9b3c"
#define CHARACTERISTIC2_UUID "4a78b8dd-a43d-46cf-9270-f6b750a717c8"

int data;
int LED_BUILTIN = 2;
bool deviceConnected = false;
int CUTOFF = -60;

class myServerCallback : public BLEServerCallbacks {

  void onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {

    char remoteAddress[18];

    sprintf(
      remoteAddress,
      "%.2X:%.2X:%.2X:%.2X:%.2X:%.2X",
      param->connect.remote_bda[0],
      param->connect.remote_bda[1],
      param->connect.remote_bda[2],
      param->connect.remote_bda[3],
      param->connect.remote_bda[4],
      param->connect.remote_bda[5]
    );

    ESP_LOGI(LOG_TAG, "myServerCallback onConnect, MAC: %s", remoteAddress);
    Serial.println("onConnect");
    Serial.println(remoteAddress);
    deviceConnected = true;
  }

  void onDisconnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {

    char remoteAddress[20];

    sprintf(
      remoteAddress,
      "%.2X:%.2X:%.2X:%.2X:%.2X:%.2X",
      param->disconnect.remote_bda[0],
      param->disconnect.remote_bda[1],
      param->disconnect.remote_bda[2],
      param->disconnect.remote_bda[3],
      param->disconnect.remote_bda[4],
      param->disconnect.remote_bda[5]
    );

    ESP_LOGI(LOG_TAG, "myServerCallback onDisconnect, MAC: %s", remoteAddress);

    deviceConnected = false;
  }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string value = pCharacteristic->getValue();
     
    if (value.length() > 0) {
        valor = "";
        for (int i = 0; i < value.length(); i++){
          // Serial.print(value[i]); // Presenta value.
          valor = valor + value[i];
        }

        Serial.println("*********");
        Serial.print("valor = ");
        Serial.println(valor); // Presenta valor.
    
        if(valor == "off") { //esto puede ser moverse adelante por ejemplo
         digitalWrite(LED_BUILTIN, LOW); 
         Serial.println("LED turned on OFF");
        pCharacteristic->setValue(alea1.c_str()); // Pone el numero aleatorio
          }
        if(valor == "on") {
          digitalWrite(LED_BUILTIN, HIGH); 
         Serial.println("LED turned on");
        pCharacteristic->setValue("LED turned on"); // Pone el numero aleatorio
          }
        if(valor == "r") {
          pCharacteristic->setValue(alea3.c_str());
         
                
        
        }//en if 
      }
    }
};
static void my_gap_event_handler(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t* param) {

  Serial.print("RSSI status");
  Serial.println(param->read_rssi_cmpl.status);
  Serial.print("RSSI ");
  Serial.println(param->read_rssi_cmpl.rssi);
  Serial.print("Address ");
  Serial.println(BLEAddress(param->read_rssi_cmpl.remote_addr).toString().c_str());
}

void setup() {
  Serial.begin(115200);
  pinMode (LED_BUILTIN, OUTPUT);//pin y su salida
  
  BLEDevice::init("Hide&Seek");// nombre del dispositivo bl
  BLEServer *pServer = BLEDevice::createServer(); // Create the BLE Server
  BLEService *pService = pServer->createService(SERVICE_UUID); 
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_NOTIFY |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pServer->setCallbacks(new myServerCallback());
  pCharacteristic->setCallbacks(new MyCallbacks());
  
  BLEDevice::setCustomGapHandler(my_gap_event_handler);
  
  pCharacteristic->setValue("Iniciado.");
  pCharacteristic->addDescriptor(new BLE2902());
  pService->start();

  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  
  pAdvertising->start();

}

void loop() {

  
  delay(1000);
  aleatorio1 = random(1,50); // Crea el numero aleatorio.
  alea1 = (String) aleatorio1; // Lo convierte en String.aleatorio
  aleatorio2 = random(100,200); // Crea el numero aleatorio.
  alea2 = (String) aleatorio2; // Lo convierte en String.aleatorio
  //alea3 = alea1 + "," + alea2;
  
//  BLEScan *scan = BLEDevice::getScan();
//  scan->setActiveScan(true);
//  BLEScanResults results = scan->start(1);
//  int best = CUTOFF;
//          Serial.print("---------NUEVO ESCANER-------------------- ");
//            Serial.print('\n');
//          for (int i = 0; i < results.getCount(); i++) {
//            BLEAdvertisedDevice device = results.getDevice(i);
//            
//            int rssi = device.getRSSI();
//            if ("Pixel 3a XL"==device.getName()) {
//              alea3=device.getName().c_str()+(String)device.getRSSI();
//              }
//            Serial.print("----------------------------- ");
//            Serial.print('\n');
//            Serial.println(device.getName().c_str());
//            Serial.println((String)device.getRSSI());
//            Serial.println(device.getAddress().toString().c_str());
//            Serial.print('\n');
//            if (rssi > best) {
//             best = rssi;
//            }//endif
//          }//endfor
     delay(15000);

TheUruguayo, 2020-10-28:
After a while I manage to do it:
class myServerCallback : public BLEServerCallbacks {//get the address for example after conection happens

    void onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {
memcpy(&peerAddress, param->connect.remote_bda, 6); //this way youll have the peerdevice  addres

 }
};
static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gap event handler, GAP event: %d", (uint8_t)event);
  
  Serial.print("RSSI from gap event"); //be carefull with the speed you ask for the rssi if you have problems
  rssi2=param->read_rssi_cmpl.rssi;
   Serial.println(rssi2);
  
}

closed 2020-10-28 by TheUruguayo

Ranajayy, 2021-05-22:
hi, i tried using this but im getting an error that peerAddress was not declared in this scope, i need to do the same thing, get the rssi of the client connected to my esp, can you please help me on this im new to the ble concept

hoangducnghi2207, 2022-04-14:
> After a while I manage to do it: class myServerCallback : public BLEServerCallbacks {//get the address for example after conection happens
>
> ```
> void onConnect(BLEServer* pServer, [...]

I also got the same error like KanaJayy, What should i do to solve this problem ?

Hoffa25, 2022-06-18:
@hoangducnghi2207 did you get it to work?

TheUruguayo, 2022-12-22:
I dont remember anything about what I did here, it was to long ago. Here I leave you with the full project so you can check it out. https://github.com/TheUruguayo/BLE-ESP32-tracking-robot
I made a robot that tracks you using BLE.
Hope it helps

TheUruguayo, 2022-12-22:
I will unsubscribe to this so I will not get further emails about it. If anyone has issues that have no answer my recomendation is to read the documentation.