nkolban/esp32-snippets #1019: micropyhon implementation of BLE mouse

street-grease-coder, 2020-10-20:
would be amazing to get your help here, bascially trying to implement your work in micropython (which would probably be insanely easy for someone who knows what they are doing...). Otherwise, please close / delete this asap I guess

https://github.com/micropython/micropython/issues/6557

chegewara, 2020-10-20:
Im not sure what do you expect, it seems you have good support from micropython devs.

street-grease-coder, 2020-10-20:
no expectations, just a hint/link to this, in case someone feels an like contributing, as said, if this is time consuming for experts, close. I'm mainly trying to understand the enormously-seeming frameworks, but if I can contribute by linking/googling... it's the minimum I can do