nkolban/esp32-snippets #1023: BLE + HTTPS(mbed) on esp32-wroom-32 clarification

jshep321, 2020-11-05:
Hi,
I have encountered the not too uncommon problems on my ESP32 board with WROOM module (4MB Flash + 520KB RAM).  Is it not possible to run BLE + a fully featured Wifi connection on this board?  I think the answer is "no".  I have other items taking up heap as well (globals, arduinojson, etc.),  Insecure wifi (http) is not an option.

I have tried the following to remedy this (after reading quite a few issues), to no avail:
1. Enable COEX
2. Disable BLE client (I'm only acting as a server)
3. Switch to nimble instead of bluedroid
4. Reduce TLS message size from default 16834 to 4800 (CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN).  Unclear if this will work for my application long term, but didn't fix it anyway.
5. Switching BLE to core 1 (expected no change and was correct :))
6. Disable BT classic (off by default in the ESP-IDF menuconfig, I believe)
7. Moving arduinojson (~400bytes) to stack

I think the conclusion here is not really the one I want--that my use case is only possible by either:
-Adding more RAM
-Swapping back and forth between BLE and full WIFI+HTTPS usage.

Am I missing something?
Thanks!

PS I'm using vscode+platformio and adafruit huzzah32 for my application.  menuconfig changes uploaded via esp-idf

chegewara, 2020-11-05:
Hi,
there is no easy answer on this question. In theory it may be possible, but all depends on full requirements and code.
I can give you few suggestions, which may help you with that:
- you are working with platformio, so, if you can try to use arduino + espidf framework; this way you can control some memory parameters in menuconfig
- try to use cJSON instead of ArduinoJSON library; i dont have proofs but i think cJSON is light weight, only problem is it is C library
- point 6) classic BT is not disabled by default in menuconfig, and you can add memory free in setup somewhere at beginning just in case
- point 4) this change will works only in case arduino + espidf, with arduino only any changes in sdkconfig are ignored
- point 3) switch to nimble is best option and should be fairly easy thanks to @h2zero who did great job porting it and making it almost 100% compatible with old library

h2zero, 2020-11-06:
Hi @jshep321, @chegewara has posted great advice already. 

Just wanted to confirm that using my NimBLE port of this repo will likely free up 40-60kB of ram. I run a web server and BLE together in my application with 100kB free heap left over.

jshep321, 2020-11-06:
Thanks guys.   I will give it a go.  I'm not clear if my earlier menuconfig updates were staying active.  I was using the CLI esp-idf to upload changes and then moving back over to vscode+platformio to upload my app.  Will this work?  

Also @h2zero, is the nimble port loaded in the default esp-idf the same as yours or do I manually pull in from here? https://github.com/h2zero/NimBLE-Arduino

h2zero, 2020-11-06:
@jshep321 If you're just using the Arduino platform setup in platformio then I suggest just adding `lib_deps = h2zero/NimBLE-Arduino` to your `platformio.ini`

jshep321, 2020-11-13:
Integrated nimble and works like a charm!  

@h2zero much thanks for the migration guide at [https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/Migration_guide.md](https://github.com/h2zero/NimBLE-Arduino/blob/master/docs/Migration_guide.md) as it was very clear!
Here's my compile time header flag you can steal if you want:
```
#ifdef BLUEDROID
    #include <BLEDevice.h>
    #include <BLEServer.h>
    #include <BLEUtils.h>
    #include <BLE2902.h>
#endif
#ifdef NIMBLE
    #include <NimBLEDevice.h>
#endif
```

closed 2020-11-13 by jshep321

h2zero, 2020-11-13:
Glad to hear it and thanks for the feedback!