nkolban/esp32-snippets #1027: ESP32 Wroom - BLEScan example. serviceUUID changeable from a Manager ?

Stargazer32584, 2020-11-13:
Hello guy's 

since last week, i tested the BLE - function on the ESP32 board.
I tried the BLEScan - example and modified it with the posibillity to scan the device and send a webhook, if the serviceUUID (send with rnf - connect for mobile is in reach.

This works fine. But i want it in modern style and configure the uuid from a Manager(ConfigManager eg.). But it is a static like:

`static BLEUUID   UUID1("88bdf79f-59ec-4f47-bf82-39b957c568a5");`

Is it possible to convert from a static to a char - string ? 

My first try was with : `static BLEUUID UUID1(ibeacon1);`

ibeacon1 is a char- string. The compilation shows no errors, but he don't react on the ibeacon1 - char.

Have anyone a suggestion ?

Thanks

chegewara, 2020-11-13:
Sure, it is possible. Dont use global static value, instead create local BLEUUID object.

Stargazer32584, 2020-11-13:
Hi,

so in this case i do this correct ?

`if (advertisedDevice.haveServiceUUID())
      {
        Serial.printf("UUID: %s ", advertisedDevice.getServiceUUID().toString().c_str());
        char BLEUUID UUID1(ibeacon1);

      }
     
      if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(UUID1) & advertisedDevice.getRSSI() > rssibeacon) { Serial.print("Korrekter Beacon gefunden")}

`
Because, he gave me a compilation error back. "Expected initializer before 'UUID1' " was the error.
Is it possible without the BLEUUID ?

Thanks a lot..

chegewara, 2020-11-13:
`BLEUUID UUID1(ibeacon1);` no char.

Stargazer32584, 2020-11-13:
Okay,

i did this with no char, but with no success.
The compiler give me the "UUID1 was not declared in this scope" - error. 
That's reproducible. 
I know, 99% of the problems are sitting in front of the monitor. 
I declared it in the local without success.

What's the problem.....?

BLE with his libary is not so easy, i think.

chegewara, 2020-11-13:
You can always try esp-idf or search for another library. I personally think it is easy to use if you know API.

Dhayu-16, 2021-06-22:
Hi friend..
 Can you guys help me, I'm having
 Problem with remoteService,, not responding or null

// Receptor o cliente de un iTag o su emulador con ESP32. Solo necesitamos la direccion MAC del pulsador (itag)

#include "BLEDevice.h"

// The remote service we wish to connect to.

static BLEUUID serviceUUID("0000ffe0-0000-1000-8000-00805f9b34");

// The characteristic of the remote service we are interested in.

static BLEUUID charUUID("0000ffe1-0000-1000-8000-00805f9b34");

static BLEAddress *pServerAddress;

static BLERemoteService* pRemoteService;

static BLERemoteCharacteristic* pRemoteCharacteristic;

const int ledPin = 2;

void setup() {

   Serial.begin(115200);
delay(10000);
   Serial.println("Starting Arduino BLE Client application...");

   BLEDevice::init("");

   delay(10000);

   

   // DIRECCION MAC DEL SERVIDOR ITAG AL QUE NOS CONECTAMOS

   

   pServerAddress = new BLEAddress("FF:FF:20:00:66:3E");

   delay(500);

   Serial.print("Forming a connection to ");

   Serial.println(pServerAddress->toString().c_str());

   BLEClient* pClient = BLEDevice::createClient();

   Serial.println("Created client");

   pClient->connect(*pServerAddress);

   Serial.println("Connected to server");

   delay(500);

     // Obtain a reference to the service we are after in the remote BLE server.

    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);

    if (pRemoteService == nullptr) {

      Serial.print("Failed to find our service UUID: ");

      Serial.println(serviceUUID.toString().c_str());

          }else{

    Serial.println(" - Found our service");}

// Obtain a reference to the characteristic in the service of the remote BLE server.

    BLERemoteCharacteristic* pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID); 
				
    if (pRemoteCharacteristic == nullptr) {

      Serial.print("Failed to find our characteristic UUID: ");

      Serial.println(charUUID.toString().c_str());

    }

    Serial.println(" - Found our characteristic");

    

}

void loop() {

  

    // Read the value of the characteristic.

    uint8_t  rxValue = pRemoteCharacteristic->readUInt8();;

     if (rxValue > 0) { 

      Serial.println(rxValue);

      delay (300);

      pRemoteCharacteristic->writeValue(0);

     }

     

   

  

delay(0);

  

  } // The End