nkolban/esp32-snippets #1031: BLE_client weird behavior with pData and Brownout

johnsmakar, 2020-12-11:
Preface: I have a Arduino Nano 33 IoT (Server), and a ESP32 NodeMCU (client).

I am trying to read my `pData` as such - 
`int data = *(int*)pData;
 Serial.println(data);`
 
but the data I am reading looks like this `756039425`
and after a couple of seconds the ESP32 triggers a brownout (regardless of the power source)

(And to confuse you even more...) after the ESP32 resets 0.1% of the time it prints the correct data before it has another brownout.

chegewara, 2020-12-11:
There is nothing confusing in brownout. Issues with power supply, usb cable or wires, period.

https://www.google.com/search?q=esp32+brownout&oq=esp32+brownout&aqs=chrome..69i57.4151j0j1&sourceid=chrome&ie=UTF-8

closed 2020-12-11 by chegewara

johnsmakar, 2020-12-12:
Thanks, I know what causes a brownout, but what is confusing is that as I said, **"regardless of the power source"** meaning usb, power bank, and lipo battery :)

Anyways, that is half of the issue, I was wondering if someone could assist me in understanding what is going on with my pData values? It still looks like it wants to print the memory address