nkolban/esp32-snippets #1033: BLERemoteService* pRemoteService = pClient->getService(serviceUUID); freezing issue

fetullah-usr, 2020-12-17:
You can find my code block in attachment. [bleClient.txt](https://github.com/nkolban/esp32-snippets/files/5709894/bleClient.txt)

1)Power on esp32
2)"We are now connected to the BLE Server." It seems product open and this code shows.(Connection succesfull)
3)Reboot esp32
4)I could not see the message again. I stucked with this following line "BLERemoteService* pRemoteService = pClient->getService(serviceUUID);"
5)Reboot product again
6)"We are now connected to the BLE Server." It seems product open and this code shows.(Connection succesfull)

This scenario continue like this for every cycle. Output logs are below.

'data: Code:1
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: , Address: 55:4c:e0:26:95:b7, manufacturer data: 4c00100507186d1dd6
BLE Advertised Device found: Name: , Address: dc:56:e7:55:b2:5d, manufacturer data: 4c0010050110bfa58f, txPower: 12
BLE Advertised Device found: Name: , Address: 29:fa:f3:6b:f3:e0, manufacturer data: 4c000906036000000000
BLE Advertised Device found: Name: , Address: 71:79:ac:ae:1a:00, manufacturer data: 4c001005471c6e32b0, txPower: 12
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc2011fb5459e8fccc5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
  Created client
onDisconnect
  Connected to server
UUDI : 4fafc2011fb5459e8fccc5c9c331914b

**REBOOTED**
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc2011fb5459e8fccc5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
  Created client
  Connected to server
UUDI : 4fafc2011fb5459e8fccc5c9c331914b
  Found our service
  Found our characteristic
The characteristic value was: Code:1
We are now connected to the BLE Server.

**REBOOTED**
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc2011fb5459e8fccc5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
  Created client
onDisconnect
  Connected to server
UUDI : 4fafc2011fb5459e8fccc5c9c331914b
'

chegewara, 2020-12-18:
I think you should first fix power issue:
> Brownout detector was triggered

When radio has not enough power supply its behavior may be unpredictable.

fetullah-usr, 2020-12-18:
I fixed power issue but my existing connection issue still Continue

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: , Address: 48:0c:7e:9f:fb:48, manufacturer data: 0600010920020331e38fc713a102d5b9c097306ea8b4c572da44eeb01a
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
 - Created client
 - Connected to server
UUDI : 4fafc201-1fb5-459e-8fcc-c5c9c331914b
 - Found our service
 - Found our characteristic
The characteristic value was: Code:1
We are now connected to the BLE Server.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: , Address: 48:0c:7e:9f:fb:48, manufacturer data: 0600010920020331e38fc713a102d5b9c097306ea8b4c572da44eeb01a
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
 - Created client
onDisconnect
 - Connected to server
UUDI : 4fafc201-1fb5-459e-8fcc-c5c9c331914b
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
 - Created client
 - Connected to server
UUDI : 4fafc201-1fb5-459e-8fcc-c5c9c331914b
 - Found our service
 - Found our characteristic
The characteristic value was: Code:1
We are now connected to the BLE Server.
data: Code:1
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
 - Created client
onDisconnect
 - Connected to server
UUDI : 4fafc201-1fb5-459e-8fcc-c5c9c331914b
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
 - Created client
 - Connected to server
UUDI : 4fafc201-1fb5-459e-8fcc-c5c9c331914b
 - Found our service
 - Found our characteristic
The characteristic value was: Code:1
We are now connected to the BLE Server.
data: Code:1
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: ESP32, Address: 24:6f:28:9d:48:06, serviceUUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, txPower: 3
Forming a connection to 24:6f:28:9d:48:06
 - Created client
onDisconnect
 - Connected to server
UUDI : 4fafc201-1fb5-459e-8fcc-c5c9c331914b

fetullah-usr, 2020-12-18:
I see the problem with different devices as well

chegewara, 2020-12-19:
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

This line suggests you still have issues with power supply. This line is only in 1 case:
- hardware reset with button or cable connect.

fetullah-usr, 2020-12-19:
I restart the device with the button to create the problem.
- connection is successful when the device is turned on.
- When I reboot again, the connection cannot be established.
- When I restart again, the connection can be established.
- This cycle continues in this way. Actually this is the problem I'm having

chegewara, 2020-12-19:
It looks like semaphore is taken, but is not released. It was fixed by few users, but never created PR if i remember.

fetullah-usr, 2020-12-19:
When I used code "BLE_server" instead of "BLE_server_multiconnect", the problem was resolved. Thank you very much for the fast support

RkVladimir, 2023-05-16:
> 

Now I have bad resolve this problem:

timer_quard = timerBegin(0, 80, true); //crutch to resolve rare BLE freezes
  timerAttachInterrupt(timer_quard, &onTimer, false);
  timerAlarmWrite(timer_quard, 1000000, false);
  timerAlarmEnable(timer_quard);

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    timerAlarmDisable(timer_quard);

but in the timer handler, when I try to send the esc32 into a deep sleep, core panic and reboot occur, which, of course, 
solves the freeze problem, but it's not good - I lose information in the RTC memory, 
tell me where in the library to correct the unlocking of the semaphore?