nkolban/esp32-snippets #1039: vscode-errors when using bluetooth (esp32)

v11, 2021-01-11:
Hello

I am using the [vscode-arduino](https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino) extension now for more than one year. Except with a few issues at beginning, i usually don't have any problem to work with my ESP32 and some other libraries.

Now, i started to work with bluetooth.

My code compiles and i could successfully test it. But: I get a lot of errors in vscode.
See: [Screenshot](https://vision11.ch/share/temp/Bildschirmfoto%202021-01-10%20um%2022.10.30.png)

This is my code:
[https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_write/BLE_write.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_write/BLE_write.ino)

Can somebody please help me? I really want to ged rid of this errors.

Thank you,
Lukas

PS: I also asked in the [vscode-arduino](https://github.com/microsoft/vscode-arduino/issues/1182) repo for some help.

This is my c_cpp_properties.json
```
{
    "configurations": [
        {
            "name": "Mac",
            "includePath": [
                "${workspaceFolder}/**",
                "/Users/vonnielu/Library/Arduino15/packages/esp32/tools/**",
                "/Users/vonnielu/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/**"
            ],
            "forcedInclude": [
                "/Users/vonnielu/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Arduino.h"
            ],
            "macFrameworkPath": [
                "/System/Library/Frameworks",
                "/Library/Frameworks"
            ],
            "intelliSenseMode": "clang-x64",
            "compilerPath": "/usr/bin/clang",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "defines": [
                "USBCON"
            ]
        }
    ],
    "version": 4
}
```

My ardunio.json
```
{
    "board": "esp32:esp32:esp32",
    "configuration": "PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=115200,DebugLevel=none",
    "sketch": "test.ino",
    "port": "/dev/tty.usbserial-0001",
    "programmer": "AVR ISP"
}
```

My settings.json
```
{
    "files.associations": {
        "__functional_base": "cpp",
        "istream": "cpp",
        "locale": "cpp",
        "memory": "cpp",
        "tuple": "cpp",
        "utility": "cpp",
        "functional": "cpp",
        "__bit_reference": "cpp",
        "array": "cpp",
        "bitset": "cpp",
        "chrono": "cpp",
        "deque": "cpp",
        "iterator": "cpp",
        "map": "cpp",
        "string": "cpp",
        "string_view": "cpp",
        "unordered_map": "cpp",
        "vector": "cpp",
        "*.tcc": "cpp"
    }
}
```

chegewara, 2021-01-11:
Just ignore those errors, its only intellisense error. You can also try to add `#include` header for std::string, i think it will be:
`#include <string>`

v11, 2021-01-11:
Thanks for the response. I understand that this is not a problem on your side. But it does not feel good to have this errors. :-)
I tried #include <string>, #include <string.h>, #include <String.h> but without success. 

I wait for some feedback from the guys at vscode-arduino. If you have additional info, why it is not working, i would be happy to know and forward it to them. But they are so far, not very responsive. 

Have a good day!

chegewara, 2021-01-11:
`#include <string>` without h extension.
It is for sure matter of proper configuration in vs code. 
One more thing you can try is to add:
`#include "Arduino.h"`

v11, 2021-01-11:
Ok it is working now with #include <string>

It only works when you have #include <string> in the code and the load the project into vscode. It does not work when you put #include <string> directly in your already opened sketch.

v11, 2021-01-11:
And thank you very much.

closed 2021-01-11 by chegewara