nkolban/esp32-snippets #1042: registerForNotify(notifyCallback) crashes with Heap Corruption

ab3lson, 2021-01-18:
I am trying to get BLE notification data from my Beetle BLE and am struggling to connect to the notify service. When using multiple BLE scanner apps on my phone, I can subscribe to the notifications and see the data flow in, but when I try to run the ESP32 scanner code, it crashes when I call `pRemoteCharacteristic->registerForNotify(notifyCallback);` I also call this function right after I call `if(pRemoteCharacteristic->canNotify())` so I know that the notify byte is set.

When the ESP32 gets to the registerForNotify line, it get this error:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c2af  PS      : 0x00060130  A0      : 0x800d23b8  A1      : 0x3ffc8610  
A2      : 0x3ffc862a  A3      : 0x00000007  A4      : 0x00000006  A5      : 0x3ffc862a  
A6      : 0x00060d20  A7      : 0x00000000  A8      : 0x800d70bd  A9      : 0x3ffc85f0  
A10     : 0x00000018  A11     : 0x3f4014a5  A12     : 0x00000018  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000007  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa  

Backtrace: 0x4000c2af:0x3ffc8610 0x400d23b5:0x3ffc8620 0x400d1c06:0x3ffc8650 0x400d1f4a:0x3ffc86a0 0x400d76dd:0x3ffc86f0 0x4008e089:0x3ffc8710
```

Here is a link to the code that I am using:

https://pastebin.com/jMAvFPj3

chegewara, 2021-01-18:
Some sensors can notify, but does not have 0x2902 descriptor. You can check with some tools like nRF connect if thats the case, also you can try aduino-esp32 master which should, not sure, handle such cases.

ab3lson, 2021-01-19:
I've tried using the BLE library, but I keep getting this error with registerForNotify() and disconnect(). Both give me a heap corruption.

chegewara, 2021-01-19:
This is to decode stack:
https://github.com/espressif/arduino-esp32#decoding-exceptions

Informations provided by you does not allow to help more.

Ce345, 2021-02-10:
@ab3lson,

Me too. Will see the degugging! To be continued.

Ce345, 2021-02-17:
Code and board is a bad combi. Have switched to html/ with js on windows 10: is functioning excellent.

Ce345, 2021-02-25:
Experimenting with several devices and software results in the conclusions:

1. Software lib ESP32 BLE Arduino works good for clients with services which have only bluetooth. 
2. Software lib ESP32 BLE Arduino works not for my smart trainer, for services with both bluetooth and ANT+ FE C.

The negative results where due to the arduuino client crashed. 
Possible direction of solution is the adaptation of the arduino software on the issue of completion of the connection setup whith a device with both ANT+ FE C & bluetooth.

Research experiments:

1. Wemos Wifi and bluetooth with battery: sketch based on BLE Server with three try outs:
a. service speed and cadence & char. csc
b. service cycle power & char. cycle power measement
c. service cycle power & char. cycle power vector

2. ESP32 dev. module: sketch based on BLE client with three try outs:

a. service speed and cadence & char. csc
b. service cycle power & char. cycle power measement
c. service cycle power & char. cycle power vector

3. Smart trainer with:
a. service speed and cadence & char. csc: bluetooth only
b. service cycle power & char. cycle power measement: ANT+ FE C & bluetooth
c. service cycle power & char. cycle power vector: ANT+ FE C & bluetooth

4. Bluetooth apps in html and javascript: ble clients 

Combining all possibilities gave positive results, except for de combinations arduino client and smart trainer for the services & char. which have both ANT+ FE C and bluetooth. Also checked  the negative results with first connecting the smart trainer ANT+ with trainer software and then with arduino client.