nkolban/esp32-snippets #1043: only 20bytes are transmitted when subscribing to a charactersitic

max763, 2021-01-19:
I can read from the esp32 more than 20bytes on client side but when I subscribe only the first 20bytes are shown. Is there a restriction within the library or do I have to search rather on client side?
regards
max

chegewara, 2021-01-19:
This is ble specific restriction which is MTU - 3 bytes.

max763, 2021-01-19:
thanks for the fast response. But should't this apply for reading the characteristic as well? But I can get more than 20bytes except when notifying the subscriber.

I've also set MTU to 92 
BLEDevice::init("ESP32_BLE");
BLEDevice::setMTU(92);

chegewara, 2021-01-19:
No, reading is MTU - 1 if i remember correctly.

setMTU is only local value set, if peer device does not support it then only default 23 bytes mtu is used.

max763, 2021-01-19:
but I have tested to read more than 60 bytes successfully with the device. only when subscribing I get not more than 20

chegewara, 2021-01-19:
This is how ble works.
When you are reading then few packets are read and merged. You dont see it, but it is how it works.
You can accept the answer or try to search with google or even in this repository. There is many similar issues/questions and answer is always the same.

max763, 2021-01-19:
ok. understood. read is merged and not bound to MTU but subscription is. thanks

chegewara, 2021-01-19:
Read also is affected by MTU, it is much slower with low MTU, just with read you will get full message.

max763, 2021-01-19:
so I have to find the reason why MTU is not increased on both sides