nkolban/esp32-snippets #1044: Esp crashs after delete pClient if there's a characteristic

SamuelNasc, 2021-01-21:
I'm trying to resolve a memory leak in my code, so I started to use `delete pClient;`, but if there's a remote characteristic envolved, like in the code below, the system crashes.

```
 pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      pClient->disconnect();
      delete pClient;
      return false;
    }
```
That's the error message:

```
 - Created client
 - Connected to server
 - Found our service
Failed to find our characteristic UUID: beb5483e-36e1-4688-b7f5-ea07361b26a8
CORRUPT HEAP: Bad head at 0x3ffe06a4. Expected 0xabba1234 got 0x3ffe0768
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400e72f7 on core 1

Backtrace: 0x40091448:0x3ffc9440 0x40091679:0x3ffc9460 0x400e72f7:0x3ffc9480 0x400910d5:0x3ffc94b0 0x40084b26:0x3ffc94d0 0x400850e1:0x3ffc94f0 0x4000bec7:0x3ffc9510 0x4008dfa7:0x3ffc9530 0x400d6c23:0x3ffc9550 0x400d49ff:0x3ffc9570 0x400d50c5:0x3ffc9590 0x400d5466:0x3ffc95b0 0x400d5485:0x3ffc95d0 0x400d3816:0x3ffc95f0 0x400d1de9:0x3ffc9620 0x400d1e6e:0x3ffc9670 0x400d7621:0x3ffc96e0 0x4008e089:0x3ffc9700

Rebooting...
```

And that's my code, just a simple alteration in the BLE Client example:

```
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("data: ");
    Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }

  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

bool connectToServer() {
    Serial.print("Forming a connection to ");
    Serial.println(myDevice->getAddress().toString().c_str());
    
    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");

    pClient->setClientCallbacks(new MyClientCallback());

    // Connect to the remove BLE Server.
    pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      pClient->disconnect();
      return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      pClient->disconnect();
      delete pClient;
      return false;
    }
    Serial.println(" - Found our characteristic");

    // Read the value of the characteristic.
    if(pRemoteCharacteristic->canRead()) {
      std::string value = pRemoteCharacteristic->readValue();
      Serial.print("The characteristic value was: ");
      Serial.println(value.c_str());
    }

    if(pRemoteCharacteristic->canNotify())
      pRemoteCharacteristic->registerForNotify(notifyCallback);

    connected = true;
}
/**
 * Scan for BLE servers and find the first one that advertises the service we are looking for.
 */
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {

      BLEDevice::getScan()->stop();
      myDevice = new BLEAdvertisedDevice(advertisedDevice);
      doConnect = true;
      doScan = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 5 seconds.
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, false);
} // End of setup.

// This is the Arduino main loop function.
void loop() {
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("We are now connected to the BLE Server.");
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    String newValue = "Time since boot: " + String(millis()/1000);
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    
    // Set the characteristic's value to be the array of bytes that is actually a string.
    pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
  }else if(doScan){
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }
  
  delay(1000); // Delay a second between loops.
} // End of loop

```

chegewara, 2021-01-21:
Did you just try to disconnect? Library should delete pClient after that and probably this is the case.

SamuelNasc, 2021-01-21:
First of all, thank you for the quick reply. And yes, i tried to disconnect, but i kept having memory leaks every time I connect with another device, so i tried to delete the pClient's

SamuelNasc, 2021-01-21:
another important thing, I can delete the pClient before obtain the characteristic, after this the system crashes

chegewara, 2021-01-21:
There is many options where the leak can be. Recently i answered on similar issue and the suggestion was to create just 1 instance of callbacks and pass only pointers.
Im not saying there is no leak in library, it is possible there is one or more. The thing is this library is no longer developed/maintained.

SamuelNasc, 2021-01-21:
Even when i don't use callbacks the leaking continues, after the fourth conexion more or less, the system stops.  Do you know another library or something?

chegewara, 2021-01-21:
Yes, Nimble library. Its using less flash and memory and is great replacement.
https://github.com/h2zero/NimBLE-Arduino

SamuelNasc, 2021-01-21:
Thank you so much @chegewara, you helped me a lot

closed 2021-01-21 by chegewara