nkolban/esp32-snippets #1045: BLE_client_passkey example doesn't work as expected

selman-nus, 2021-01-22:
Hi,

I am trying to connect a BLE device that has a pin code. I tried [BLE_client_passkey.ino](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_client/BLE_client_passkey/BLE_client_passkey.ino) example and searched in closed issues here but still couldn't figure out the problem. 

Is this example supposed to function properly? If not, is there any other code snippet that I can try? 

Thank you!

chegewara, 2021-01-25:
This example is just an example, but there is few possible security options and you have to change it to match your needs, which depends on peer device.

selman-nus, 2021-01-26:
Hi @chegewara thanks for your reply. The peer device has a PIN code that needs to be entered to enable the connection. I know the PIN and I am able to use it to read data with my mobile phone using nrf connect app. 

So yes I changed the example code to match the PIN code with my peer device's PIN code. You can see my complete code below. It seems security callbacks are not even called once. 

What am I missing? What security options are you referring to? I can provide more details on the peer device if necessary. 

[https://pastebin.com/zgjgTngS](https://pastebin.com/zgjgTngS)

chegewara, 2021-01-26:
Maybe this function, not necessary with that parameter value:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/security/SampleServer_Encryption.cpp#L43

selman-nus, 2021-01-27:
Thank you for the suggestion. I tried different parameter values based on [this](https://github.com/espressif/arduino-esp32/blob/2452c1fb539246e47a715b74a3ad25b8a7ebbec7/tools/sdk/include/bt/esp_gap_ble_api.h#L251) header file. 

So options are : 
- ESP_BLE_SEC_ENCRYPT_MITM 
- ESP_BLE_SEC_ENCRYPT_NO_MITM  
- ESP_BLE_SEC_ENCRYPT 

However, the outcome is the same. ESP32 is not able to proceed with the connection using PIN code. Here is the serial monitor output:

```
21:12:46.813 -> Starting Arduino BLE Client application...
21:12:47.516 -> BLE Advertised Device found: Name: BLEsmart_0000011AB0495F076F85, Address: b0:49:5f:07:6f:85, serviceUUID: ecbe3980-c9a2-11e1-b1bd-0002a5d5c51b, txPower: 0
21:12:47.516 -> Forming a connection to b0:49:5f:07:6f:85
21:12:47.516 ->  - Created client
21:12:47.516 ->  - Connected to server
21:12:47.610 -> onDisconnect
21:12:47.610 -> Failed to find our service UUID: 00001810-0000-1000-8000-00805f9b34fb
21:12:47.610 -> We have failed to connect to the server; there is nothin more we will do.
```

It says failed to find our service UUID but the device clearly has that UUID. Otherwise, the device wouldn't found while scanning for this service UUID in the first place. And of course, I checked it twice using nrf connect app. UUID is correct. 

Do you have any other suggestions?

chegewara, 2021-01-27:
I would add custom gap and gatc callbacks. You can find more info about it in few issues and BLEDevice.