nkolban/esp32-snippets #1046: Unable to read data from cycling trainer

jimmy142, 2021-01-22:
Hi everyone, 

I hope someone can help me out trying to create a BLE client to read the power measurement on my cycling trainer! I'm pretty new to all this but it's driving me crazy for days, it just doesn't seem to work, I think I must be missing something obvious...

Using nRF connect app, I can connect to the device. It has three advertised services:
Fitness machine, UUID 1826; Cycling Power, UUID 1818; and Cycling Speed and Cadence, UUID 1816. 

Looking in the attribute table under UUID 1818, there's the characteristic 'Cycling Power Measurement' with a UUID 2A63. This is what I would like to read, as I pedal I can see the numbers change.

So to implement this on the ESP32, in my mind I need to connect to service 1818, and look for characteristic 2A63. When I run the client sketch with the following:

static BLEUUID serviceUUID("00001818-0000-1000-8000-00805f9b34fb");
static BLEUUID    charUUID("00002A63-0000-1000-8000-00805f9b34fb");

I get the following:
Forming a connection to c3:dd:e5:7f:d7:d5
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
We are now connected to the BLE Server.
Setting new characteristic value to "Time since boot: 3"
Notify callback for characteristic 00002a63-0000-1000-8000-00805f9b34fb of data length 15
data: 1

The data is not what I expected, nRF shows the value of this (when not pedaling) is 0x31-00-00-00-4C-92-A5-05-00-8E-2E-41-B9-FD-0B. Any ideas what I'm missing??

Much obliged,
Jim

jimmy142, 2021-01-23:
Apologies for the double post, but I thought I would update to say I've now managed it. I think writing the original post helped clarify my mind. The 'data length 15' should have told me the data is there, I just wasn't displaying it properly. Taking the second and third byte and doing the following enabled me to get the power in Watts: 

    RP2 = pData[2];
    RP3 = pData[3];
    TPWR = (RP2 + (RP3 * 256));

Now on to the next challenge (actually there are two). Firstly, I only manage to connect to the server ~50% of the time, I simply have to keep restarting the ESP32 until it works, nothing else changes, which is very weird behaviour to me and difficult to resolve as it is intermittent... any ideas? Secondly, I'd like to also read the cadence value from the same cycle trainer, it has a service UUID 1816 and characteristic UUID 2A5B, but I've no idea how to modify the code to do this, I even thought about using a second ESP32 but I'm not sure you can make two BLE connections to the cycle trainer at the same time...  Any pointers?

Cheers,

Jim

chegewara, 2021-01-25:
Thats good you solved problem.
Now you want to read another value, so duplicate the code to find characteristic and register for notify (assuming the characteristic is sending notifications). You can use the same notify callback or have another one to simplify it.

Connection issue can be related to weak signal or maybe you need to wait a bit longer before reconnect.

cfprogrammers, 2021-02-08:
jimmy142

cfprogrammers, 2021-02-08:
jimmy142 ,

You referred to the client sketch in your post. I cannot find the sketch in the esp32-snippets folder. Please provide the link to the sketch, I have the similar problem with ESP32 client

Ce345, 2021-02-10:
jimmy142,

For your challenge (off bike!):
"UUID 1816 and characteristic UUID 2A5B, but I've no idea how to modify the code to do this."
Anser:
Just replace 1818 by 1816 in line with declaration:  static BLEUUID serviceUUID("00001818-0000-1000-8000-00805f9b34fb"); 
and replace 2a63 by 2a5b in line with declaration: static BLEUUID    charUUID("00002a63-0000-1000-8000-00805f9b34fb");

Question to you:
Which physical board and board-definition do you use?
Did you use BLE_client from ESP32 BLE Arduino and which version of the lib?
Reason for this is a continuous crash due to heapsize problems of my board when trying to get the notify for the powermeasurement ( "2a63"). A new issue is also considerd, of course.

jimmy142, 2021-02-13:
Hi Ce345,

Thanks, I haven't made any progress for a while (no time), but looking at my setup a bit more closely I see there's a separate cadence sensor so I need to figure out how to read the power from the trainer and then the cadence from the other device. I guess it's just a case of playing around.

In terms of the board, I'm using a ESP32-DevKitC with ESP32-WROOM-32D, made by AITRIP (amazon purchase). Board-definition is the ESP32 DEV MODULE in Arduino IDE. Will need to check my laptop to find library version....

@cfprogrammers, sure, I'll put up a link to the sketch, give me a couple of days.

Ce345, 2021-02-17:
The board seems nothing special.
Succeeded today once in getting the data.
Dropped this time-consuming excersise. 
Have already made (always!) working webpages, with html and javascript, for heartrate, power and cadence & speed from BLE devices. 
Succes with the attempts.

Ce345, 2021-02-25:
Experimenting with several devices and software results in the conclusions:

Software lib ESP32 BLE Arduino works good for clients with services which have only bluetooth.
Software lib ESP32 BLE Arduino works not for my smart trainer, for services with both bluetooth and ANT+ FE C.
The negative results where due to the arduuino client crashed.
Possible direction of solution is the adaptation of the arduino software on the issue of completion of the connection setup whith a device with both ANT+ FE C & bluetooth.

Research experiments:

Wemos Wifi and bluetooth with battery: sketch based on BLE Server with three try outs:
a. service speed and cadence & char. csc
b. service cycle power & char. cycle power measement
c. service cycle power & char. cycle power vector

ESP32 dev. module: sketch based on BLE client with three try outs:

a. service speed and cadence & char. csc
b. service cycle power & char. cycle power measement
c. service cycle power & char. cycle power vector

Smart trainer with:
a. service speed and cadence & char. csc: bluetooth only
b. service cycle power & char. cycle power measement: ANT+ FE C & bluetooth
c. service cycle power & char. cycle power vector: ANT+ FE C & bluetooth

Bluetooth apps in html and javascript: ble clients

Combining all possibilities gave positive results, except for de combinations arduino client and smart trainer for the services & char. which have both ANT+ FE C and bluetooth. Also checked the negative results with first connecting the smart trainer ANT+ with trainer software and then with arduino client.