nkolban/esp32-snippets #1049: BLE Memory Leak

nnangeroni, 2021-02-21:
Dear readers,

I'm struggling with a memory leak of 20 bytes that happens during every BLE access.  I've spent some time with issue #963 which is very helpful but when I change to pointers for advertisedDevice as suggested I get the error 

>   invalid new-expression of abstract class type 'BLEdeviceCallbacks'

The offending code is   

>   pBLEScan->setAdvertisedDeviceCallbacks(new BLEdeviceCallbacks);

but I am not very experienced with C++ or BLE and haven't been able to figure out how to replace this code to be able to use the fix for memory leak.

My code follows.

Thank you for any help.
Best regards,
Nancy

`#include "BLEDevice.h"
#include <BLEAdvertisedDevice.h>

#define OFF 0
#define RUNNING 1
byte autoState = OFF;
#define BLE_OFF 0
#define BLE_SCANNING 1
#define BLE_BEACON 2
static byte BLEstate = BLE_OFF;
float newTemp = 0.0;
unsigned long time4TempSample = 0;
unsigned long delaySample = 29500;
boolean gotTemperature = false;
static BLEAdvertisementData advertisementData;
BLEScan* pBLEScan;
unsigned long time2ScanEnd = 0;
#define SCAN_LENGTH_s     10      // pBLEscan time in seconds
#define SCAN_INTERVAL    603
#define SCAN_WINDOW      549
#define SCAN_LENGTH_ms  SCAN_LENGTH_s*1000

void    BLEunpackMeeblueAdv(char* payl) {
  newTemp = (payl[22] << 8) + payl[21]; 
  newTemp = ((175.72*newTemp)/65536.0) - 46.85;
  newTemp = (newTemp * 1.80) + 32.00 ;
}
class   BLEdeviceCallbacks: public BLEAdvertisedDeviceCallbacks { 
  void onResult(BLEAdvertisedDevice* advertisedDevice) {
    char* payl = (char*)advertisedDevice->getPayload();
    String payload = String(payl);
    size_t payloadLength = advertisedDevice->getPayloadLength();
    byte targetMeeblue[] = {2,1,6,3,3,0};
    boolean foundMeeblue = true;
    for (byte j=0; j < 6; j++) {
      if (payl[j] != targetMeeblue[j]) foundMeeblue = false;
      if (!foundMeeblue) j = 7;
    }
    if (foundMeeblue) {
      time2ScanEnd = millis();
      gotTemperature = true;
      BLEunpackMeeblueAdv(payl);     // unpack temp, humidity and battery
      BLEstate = BLE_BEACON;
    } 
  }
};
void    BLEperformScan() {    // BLOCKING
  time2ScanEnd = millis() + (SCAN_LENGTH_s * 1000);
  BLEstate = BLE_SCANNING;
  BLEDevice::init("");    // for MANUAL startup, when no BLEinit
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new BLEdeviceCallbacks);
  pBLEScan->setInterval(SCAN_INTERVAL);
  pBLEScan->setWindow(SCAN_WINDOW);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(SCAN_LENGTH_s, false);
  pBLEScan->stop();
  pBLEScan->clearResults();
}
void setup() {
  Serial.begin(57600);
  Serial.println("\nStarting up");
  BLEstate = BLE_OFF;
  esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);  // release memory not needed for BLE
}

void loop() {
  if (autoState == OFF) {
    if (gotTemperature) autoState = RUNNING;  
  }
  if (BLEstate == BLE_SCANNING) { 
    if (millis() > time2ScanEnd) BLEstate = BLE_OFF;
  }
  else if ((BLEstate == BLE_OFF)) 
    BLEperformScan();  
  else if (BLEstate == BLE_BEACON) {   
    if (millis() > time4TempSample) {
      if (gotTemperature) {
        Serial.print(String(newTemp));
        Serial.println("  Heap: " + String(ESP.getFreeHeap()));  
        time4TempSample = millis() + delaySample;  // setup next temp sample
        gotTemperature = false;
      }
      else  {
        BLEstate = BLE_OFF;
        autoState = OFF;
      }
    }
  }
}`

chegewara, 2021-02-21:
Recently some memory leak fix has been merged in this library clone in aruino-esp32 repository.

nnangeroni, 2021-02-21:
Thank you! I'm using version 1.0.1 of ESP32 BLE Arduino, the library manager doesn't show any more recent version. The repositories I can find are 2 years old, can you give me a link?

Thanks, sorry such newbie.
Nancy

nnangeroni, 2021-02-21:
I updated BLEDevice and BLEAdvertisedDevice src files with version dated 13 months ago, I'm still seeing precisely 20 bytes loss every iteration.

chegewara, 2021-02-21:
https://github.com/espressif/arduino-esp32/pull/4761

But im not sure if it really is fixing all leaks, didnt test it.

nnangeroni, 2021-02-21:
Fix is for leak when getting characteristics. I'm using a beacon, getting data from advertising packets.

nnangeroni, 2021-03-07:
I have solved this problem, by moving the following from BLEperformScan() to Setup():

    BLEDevice::init("");
    pBLEScan = BLEDevice::getScan();
    pBLEScan->setAdvertisedDeviceCallbacks(new BLEdeviceCallbacks(), false); 
    pBLEScan->setActiveScan(true);
    pBLEScan->setInterval(SCAN_INTERVAL);
    pBLEScan->setWindow(SCAN_WINDOW);    

This way the 'new' call is happening only once.

closed 2021-03-07 by nnangeroni