nkolban/esp32-snippets #1050: ESP32 BLE onDisconnect() gets called after some delay

JLightMedia, 2021-02-23:
I use a standard BLE client example. After a successful connection, a program example sends some data to a server. In case of disconnection of the server, the client detects that and calls onDisconnect()  only after some delay (about 6 sec). But the main loop continues to send data. And terminal just sends an error "lld_pdu_get_tx_flush_nb HCI packet count mismatch (10, 11)" and after the board can work again in the normal way. 
But if the delay between data sending less than 200 milliseconds, the terminal just sends "onDisconnect" message, and the bord reboots or just stops working. 
I equate understand, what happens, as I try to send data to Client that has already offline and some issue leads to that. But I don't understand why onDisconnect() method is called with that delay? I've tested this code on different ESP32 board with ArduinoIDE, but the result always the same.

Petros144, 2021-04-21:
Hi @JLightMedia I have similar Errors with my iPhone and ANCS Service, if I go out of range from the ESP I get a wierd behaviour of rebooting or Stuck of my program... I also use Arduino IDE. I think it is the same bug as you have.

I have no clue how to tackle this problem...

I also find this Post on ESP forum with similar problem:

https://esp32.com/viewtopic.php?t=7445

cheers

joaovmt, 2021-07-16:
Hello guys, have some update about this? I have the same problem in Arduino IDE. When i get out of server alcance with client (ESP32) and return, i have this error. 16:47:58.823 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)