nkolban/esp32-snippets #1055: Problems receiving Broadcast Advertisements by Windows

Intrpt, 2021-03-09:
Hi guys,

I'm using my ESP32 to receive Broadcast Advertisements of my Laptop.
My Problem is that I don't receive all of the Advertisements.
Sometimes I don't receive them at all.
I also used my Linux Laptop to receive the Advertisements and it is working. This is why I try to find the issue on the ESP32.

My Scan interval and Scan window on my ESP is set to 100. So I guess it should scan all the time.
I tried to scan for 1 Second 12 times and I tried to scan for 12 seconds in hope it will make any difference, but it does not.

Right now I don't have any ideas what I could try, I'm hoping that you guys could give my some new ideas what I could try?

chegewara, 2021-03-09:
Since you can find you linux laptop then all is ok with esp32 and code. You seem to have problem with windows app that is advertising. My advise is to use nRF connect app, best on android, but iPhone also is fine, to diagnose and check if you can find windows laptop.

Intrpt, 2021-03-09:
What I mean is I used my Linux Laptop and my ESP32 to receive the Advertisement Broadcasts from my Windows Laptop.
With Linux it is working all the time. With ESP32 it is not. This is why I feel like there must be sth. wrong with my ESP32 settings.

I used nRF Connect on my Android phone, it is also working. 
The following is the result from the nRF app:

Advertising type: Legacy
Scenario Type: Advertising Beacon <0x01>
Version: 0
Device Type: Windws 10 Desktop
Flags: 0x00
Salt:0x9DC278C1
Device Hash: [...]
RSSI: -60dBm
Adv. Interval: 100ms

chegewara, 2021-03-09:
What board are you using. It may be hardware issue or external antenna is required.

Intrpt, 2021-03-09:
I'm using SBC-NODEMCU-ESP32

right now I'm trying to use my own custom gap_event_handler insead of BLEAdvertisingCallback. It looks like I receive all Broadcasts within this GAP Event Handler.

closed 2021-03-09 by Intrpt