nkolban/esp32-snippets #1057: Can't count devices (foundDevices.getCount() doesn't work

hendrijunianto, 2021-03-20:
Please help me, I do scanning for Bluetooth device with my esp32, as seen in the picture there are 2 devices founded but why the "device found" count still 0? Thanks in advance.
[image]
[image]

Subcode, 2021-06-26:
I'm getting the same problem.

hendrijunianto, 2021-06-26:
Down grade your board to version 1.0.4, it works for me

Subcode, 2021-06-26:
You cant use AdvertisedDeviceCallbacks and getCount at the same time.
Either dont use getCount, or dont use the callback.