nkolban/esp32-snippets #1068: Choose BLEUUID service at launch

eqcq, 2021-04-10:
Hi, and thanks a lot for all the work.
Is it possible to choose a service before start the search ?

In your client exemple you use : BLEUUID service("uuid");
how can for exemple create an array or a struct in order to choose before launching wich service will be scan and connect ?

is it possible to change this service without reset my device ?

thx for all

chegewara, 2021-04-10:
Im not sure i understand your needs, can you share part of code you are asking about?

chegewara, 2021-04-13:
This is just a simple example. Your peripheral device, you want to interact with, may have many services and characteristics.
I UART example you have 2 chars:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_uart/BLE_uart.ino

You have to improve code to use all UUIDs you need.

eqcq, 2021-04-13:
Sorry for the misunderstanding but I known that I could create different services and characteristic. 
I just want to select dynamically at launch one of them (with a roto encoder) by using an array for exemple, and I couldn't exactly determine the data types of UUIDin order to declare my array or my structure.
finally I understand how to achieve this. I have declared all my UUIDS like this :

static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b »);

and create my array after like this :

BLEUUID services[]={serviceUUID,servicesTWOUUID, ...};

sometime a bit of sleep make the things easier … 
thx for the support and the library

Guillaume 

www.errorishuman.com

chegewara, 2021-04-13:
Yes, yhat is good way to do it.