nkolban/esp32-snippets #1069: How to disconnect properly a BLE client from a server?

eqcq, 2021-04-14:
Hi there :)

I'm trying to disconnect my ble client from it server in order to connect it to another server.
I would like to achieve this in the client sketch. I've found  Client->disconnect(); but it makes my esp32 reboot.
The idea is to choose in a menu which service is used by my client device.

here is my code for the ble part. there's a bluetooth function called in setup().

`  BLEClient* octoBLE;
//////////////////////////////////////////////////////////////////////
//            displaysay scan interface                             //
//////////////////////////////////////////////////////////////////////
void go() {
  encoderValue = 0;
  Ecran.fillScreen(theme[themecolor].fonds);
  resetRepereColor();
  DrawBasicInterface();
  rotaryEncoder.setBoundaries(bornage[0][0], bornage[0][1], false);
  menu = Mhome;
}
//////////////////////////////////////////////////////////////////////
//        Variables necessaires                                     //
//////////////////////////////////////////////////////////////////////
static boolean doConnect = false;
static boolean paired = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;
//////ticker
Ticker ListenBLE;

//////////////////////////////////////////////////////////////////////
//            BLE CALLBACKS                                         //
//////////////////////////////////////////////////////////////////////
///////////reception notifications
static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
  std::string myString((char*)pData, length);
  ChaineMemoire = myString.c_str();
}
///////////ecouteur connected / deconnexion
class MyClientCallback : public BLEClientCallbacks {
    void onConnect(BLEClient* pclient) {
    }
    void onDisconnect(BLEClient* pclient) {
      DrawInit("Disconnected");
      playstate = Sstop;
      StopAll();
      Etat = "Stop";
      paired = false;
    }
};
//////////////////////////////////////////////////////////////////////
//            CONNEXION LISTENER                                    //
//////////////////////////////////////////////////////////////////////
bool connectToServer(BLEClient* pClient) {
  DrawInit("Connexion");
  pClient  = BLEDevice::createClient();
  pClient->setClientCallbacks(new MyClientCallback());
  pClient->connect(myDevice);
  BLERemoteService* pRemoteService = pClient->getService(bledevices[WhatDevice].service);
  if (pRemoteService == nullptr) {
    pClient->disconnect();
    return false;
  }
  pRemoteCharacteristic = pRemoteService->getCharacteristic(bledevices[WhatDevice].charact);
  if (pRemoteCharacteristic == nullptr) {
    pClient->disconnect();
    return false;
  }
  if (pRemoteCharacteristic->canRead()) {
    std::string value = pRemoteCharacteristic->readValue();
  }
  if (pRemoteCharacteristic->canNotify())
    pRemoteCharacteristic->registerForNotify(notifyCallback);
  go();
  paired = true;
  return true;
  
}
//////////////////////////////////////////////////////////////////////
//            SCAN SERVERS CLASS                                        //
//////////////////////////////////////////////////////////////////////
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(bledevices[WhatDevice].service)) {
        BLEDevice::getScan()->stop();
        myDevice = new BLEAdvertisedDevice(advertisedDevice);
        doConnect = true;
        doScan = true;
      }
    }
};
//////////////////////////////////////////////////////////////////////
//            ECOUTEUR                                              //
//////////////////////////////////////////////////////////////////////
void watchBLE() {
  if (doConnect == true) {
    if (connectToServer(octoBLE)) {
      //connected to server
      doConnect = false;
    } else {
      //fail
      doConnect = true;
    }
  }
  if (!paired && doScan) {
    DrawInit("Scanning BLE");
    BLEDevice::getScan()->start(0);
  }
}
////////////////////////////////////////////////////////////////////////////////////Initialisation
void bluetooth() {
  BLEDevice::init("OCTO");
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(0);
  DrawInit("Scanning BLE");
  watchBLE();
  ListenBLE.attach_ms(1000, watchBLE);
}`

CLDiego, 2021-04-14:
Having a look at your code I could not found an instance where you intentionally trigger the disconnect. The only disconnect instances are within the connection process, so it's hard to tell what could be the problem.

I've used the library before to connect and disconnect to multiple servers, and haven't got an issue with the disconnect method.  

One thing that might help with your issue would be to track where the restart occurs. The disconnect method should trigger the onDisconnect callback.

eqcq, 2021-04-14:
There’s still no disconect in my code
I juste do something like bleclientname->disconnect(); and it crushes

Guillaume Pascale
438 630 7724
www.errorishuman.com