nkolban/esp32-snippets #1074: Connect to BLE server without scanning

Murtaza7745, 2021-04-24:
I am new to ESP32 BLE programming, I want to connect ESP32 BLE client to only one particular server, I have the server ble address and its charac UUID and service UUID.

PLEASE HELP

chegewara, 2021-04-24:
https://github.com/nkolban/esp32-snippets/issues/1072#issuecomment-825633790

Murtaza7745, 2021-04-24:
Thanks so much for replying so fast @chegewara.
Let me explain you in a different way,
I have a esp32 BLE acting as a client, i have a smart ble watch which is acting as a server.
I want the client to connect to that server only, without scanning or anything, just use "static BLEClient*pClient->connect("60:35:0d:6c:79:21");"

PLEASE HELP

chegewara, 2021-04-24:
In post i linked you can read its not possible. You have to scan.

Murtaza7745, 2021-04-24:
so how do I increase the memory size? if there are so many ble scan results, esp32 reboots.

chegewara, 2021-04-24:
There is few options.
- use nimble library instead
- perform short scan with custom interval and window

Murtaza7745, 2021-04-24:
Thanks @chegewara :)