nkolban/esp32-snippets #1075: Unstable bluetooth connection

cvanelteren, 2021-04-29:
Hi,

Thanks for the project. After running some test code I am yielding an unstable connection that keeps disconnecting
This is the main error that is posted every few seconds : 

`lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)` 
In addition, I can see in my bluetooth manager that the HID is disconnecting and reconnecting.
Are there any pointers for this issue?
Best,
C

chegewara, 2021-04-29:
Hi,
maybe it is due to weak signal. Did you try to increase tx power?
There is few issues here explaining how to do it.

cvanelteren, 2021-04-29:
Thanks for the quick reply. Oddly enough a removal from my bluetooth client made the instability go away :S

closed 2021-04-29 by cvanelteren