nkolban/esp32-snippets #1081: Esp32 ble. When the connection is lost, it does not connect again, I need to reset it. Can you help me with this?

paxsipornax, 2021-05-23:
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLECharacteristic *pCharacteristic;
bool deviceConnected = false;

#define SERVICE_UUID "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class MyServerCallbacks: public BLEServerCallbacks {
void onConnect(BLEServer* pServer) {
deviceConnected = true;
};

void onDisconnect(BLEServer* pServer) {
  deviceConnected = false;
}
};

class MyCallbacks: public BLECharacteristicCallbacks {
void onWrite(BLECharacteristic *pCharacteristic) {
std::string rxValue = pCharacteristic->getValue();

  if (rxValue.length() > 0) {
    Serial.println("*********");
    Serial.print("Received Value: ");

    for (int i = 0; i < rxValue.length(); i++) {
      Serial.print(rxValue[i]);
    }

    Serial.println();
  }
}    
};

void setup() {
Serial.begin(115200);

BLEDevice::init("ESP32 UART Test");

BLEServer *pServer = BLEDevice::createServer();
pServer->setCallbacks(new MyServerCallbacks());

BLEService *pService = pServer->createService(SERVICE_UUID);

pCharacteristic = pService->createCharacteristic(
CHARACTERISTIC_UUID_TX,
BLECharacteristic::PROPERTY_NOTIFY
);

pCharacteristic->addDescriptor(new BLE2902());

BLECharacteristic *pCharacteristic = pService->createCharacteristic(
CHARACTERISTIC_UUID_RX,
BLECharacteristic::PROPERTY_WRITE
);

pCharacteristic->setCallbacks(new MyCallbacks());

pService->start();

pServer->getAdvertising()->start();

}

void loop() {
if (deviceConnected) {

}
delay(1000);
}

chegewara, 2021-05-23:
You have to restart advertising in `onDisconnect()`.

paxsipornax, 2021-05-23:
> İçinde reklam vermeye yeniden başlamalısınız `onDisconnect()`.

   void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
  

    pServer->getAdvertising()->start();
      
    }
};

it worked, it worked thanks 👍

closed 2021-05-24 by chegewara