nkolban/esp32-snippets #1083: Is it possible to change the duration of the scan from uint32_t to float?

reveriess, 2021-06-22:
i need the BLE to scan and re-scan in less than 1 second interval. i tried to change the data type of duration from uint32_t to float. it can be compiled and uploaded to my ESP32 WROOM devkit v1 but now it couldn't scan anything. 

is there another way to do this? or does the scantime is limited to integer seconds only? thanks

chegewara, 2021-06-22:
No, it is not possible. It is how esp-idf API is designed.

adojang, 2021-09-29:
Hi there, apologies for being a few months late to the party. Is there an alternative to using this library?
Under https://github.com/nkolban/ESP32_BLE_Arduino The author states that BLE is now directly included within the Arduino, but I haven't been able to find good documentation on how to use the library.

@reveriess This post here https://www.esp32.com/viewtopic.php?t=7615 mentions that:

Inside your `void onResult` function
at the beginning, if you write:
`advertisedDevice.getScan()->stop();`

It will stop the currently ongoing scan. Might not be a perfect solution, but its a temporary work around for me. 
Cheers @chegewara :)