nkolban/esp32-snippets #1084: ESP32 BLE multiple UUIDs for notifications

xActTM, 2021-06-23:
Hello, I've been trying to use my ESP32 as a WiFi remote for my GoPro MAX to automate bracketed photos. Unfortunately, GoPro implemented something causing WiFi to be shut off until something connects to it's BLE signal and sends a command telling it to enable WiFi. I've used the BLE-client example to connect to it, and it says through serial that it successfully connected, but the GoPro doesn't say it connected. GoPro's "Open GoPro" GitHub says this is because the device must "Subscribe to notifications from all characteristics. The BLE_client example seems to have code showing that it accepts notifications from the supplied characteristic if it supports it, but I'm unable to figure out how it works, and how to do it with multiple UUID's (from what I see on GoPro's GitHub there are three of them) This is my first Bluetooth/ESP32 related project, so sorry if I missed something obvious.

Here's GoPro's Open GoPro GitHub: https://gopro.github.io/OpenGoPro/

And here's my code:
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

// GoPro Service UUID
static BLEUUID serviceUUID("0000fea6-0000-1000-8000-00805f9b34fb");
// GoPro Characteristic UUID
static BLEUUID    charUUID("b5f90077-aa8d-11e3-9046-0002a5d5c51b");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
  Serial.print("data: ");
  Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
    void onConnect(BLEClient* pclient) {
    }

    void onDisconnect(BLEClient* pclient) {
      connected = false;
      Serial.println("onDisconnect");
      digitalWrite(2, LOW);
    }
};

bool connectToServer() {
  Serial.print("Forming a connection to ");
  Serial.println(myDevice->getAddress().toString().c_str());

  BLEClient*  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");

  pClient->setClientCallbacks(new MyClientCallback());

  // Connect to the GoPro's BLE server
  pClient->connect(myDevice);  // If you pass BLEAdvertisedDevice instead of the GoPro address, it will be recognized as a type of peer device address (public or private)
  Serial.println(" - Connected to server");

  // Obtain a reference to the server
  BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    Serial.println(serviceUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println("Server found");

  // Obtain a reference to the characteristic in the server
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(charUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our characteristic");

  // Read the characteristic's value
  if (pRemoteCharacteristic->canRead()) {
    std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());
  }

  if (pRemoteCharacteristic->canNotify())
    pRemoteCharacteristic->registerForNotify(notifyCallback);

  connected = true;
  return true;
}
/*
   Scan for BLE servers and find the one that matches the GoPro
*/
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    /**
        Called for each advertising BLE server.
    */
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.print("BLE Advertised Device found: ");
      Serial.println(advertisedDevice.toString().c_str());

      // Found the device
      if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {

        BLEDevice::getScan()->stop();
        myDevice = new BLEAdvertisedDevice(advertisedDevice);
        doConnect = true;
        doScan = true;

      } // Found the server
    } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  pinMode(2, OUTPUT);
  BLEDevice::init("");
  /*
    Retrieve a scan and set the callback we want to use to be informed when it
    has detected a new device. Specify that we want active scanning and start the
    scan to run for 5 seconds.
  */
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, false);
} // End of setup.

void loop() {
  /*
    If the flag "doConnect" is true then it has found the desired
    BLE Server with which we wish to connect. Once we are
    connected it will set the connected flag to be true.
  */
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("Now connected to the GoPro BLE Server.");
      delay(1000);

      uint8_t array[4] = {03, 17, 01, 01};
      pRemoteCharacteristic->writeValue(array, 4);
      Serial.println("Wrote Value(s) with a size of:");
      Serial.println(sizeof(array));

      digitalWrite(2, HIGH);
    } else {
      Serial.println("Failed to connect to BLE server");
    }
    doConnect = false;
  }

  delay(1000); // Delay between next search
}
```

chegewara, 2021-06-24:
In this example you have only 1 characteristic. Now you have to duplicate part of code for all characteristics you want to use:
```
  // Obtain a reference to the characteristic in the server
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(charUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our characteristic");

  if (pRemoteCharacteristic->canNotify())
    pRemoteCharacteristic->registerForNotify(notifyCallback);
```
You can have the same `notifyCallback` for all characteristics, but it will be much easier to have 1 per characteristic.

xActTM, 2021-06-24:
I tried duplicating that part of the code and changing the charUUID to be each characteristic value, but it still doesn't work. Am I doing this wrong?

```
  // Obtain a reference to the characteristic in the server
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(charUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our characteristic");
  Serial.println(charUUID.toString().c_str());

  if (pRemoteCharacteristic->canNotify())
    pRemoteCharacteristic->registerForNotify(notifyCallback);
    
  // Obtain a reference to the characteristic in the server
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID2);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(charUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our characteristic");
  Serial.println(charUUID2.toString().c_str());

  if (pRemoteCharacteristic->canNotify())
    pRemoteCharacteristic->registerForNotify(notifyCallback);

      // Obtain a reference to the characteristic in the server
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID3);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(charUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our characteristic");
  Serial.println(charUUID3.toString().c_str());

  if (pRemoteCharacteristic->canNotify())
    pRemoteCharacteristic->registerForNotify(notifyCallback);
```

chegewara, 2021-06-24:
Here everything looks food, just you have to create separate variable:
`static BLERemoteCharacteristic* pRemoteCharacteristic2;`
which will be the one that you can write to, or even 3 more if you want to write to all 3 characteristics.

xActTM, 2021-06-27:
I don't know if I'm doing something wrong, but it just won't work. I tried creating a separate "pRemoteCharacteristic" for each characteristic but that just makes it crash upon connecting to the GoPro... I tried it all using the same variable and it connects, but the camera still says it didn't connect.

[This person](https://github.com/KonradIT/goprowifihack/issues/191) might've had success by doing a scan first. I don't know if they mean scanning for devices or something else but the code I'm using should already do that right?

chegewara, 2021-06-27:
Yes, you do scan anyway.
I have read GoPro docs and it is all nicely described and easy to implement, just you have to know what you do in arduino/C++ code.

xActTM, 2021-06-27:
I agree, but I am fairly new to Arduino IDE and this is my first Bluetooth project involving the ESP32. I'm not sure what else to try at this point as everything seems to be okay, the GoPro just refuses to show that it connected. The only thing I can think of is maybe something is wrong with the part telling it to receive notifications:

```
  // Obtain a reference to the characteristic in the server
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(charUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our characteristic");
  Serial.println(charUUID.toString().c_str());
```
Or maybe the way the pRemoteCharacteristic variable is set up:
```
static BLERemoteCharacteristic* pRemoteCharacteristic;
```

chegewara, 2021-06-27:
For example, to enable AP (it is what you are trying to do), you have to write to characteristic with UUID:
`b5f90072-aa8d-11e3-9046-0002a5d5c51b`
and you will get notification on characteristic with UUID:
`b5f90073-aa8d-11e3-9046-0002a5d5c51b`

At the moment youare searching only for characteristics with notifications and in loop you also are trying to write to characteristic with notify. You have to add code to add some code:
```
static BLEUUID    charUUID("b5f90072-aa8d-11e3-9046-0002a5d5c51b");
  pCmdCharacteristic = pRemoteService->getCharacteristic(cmdUUID);
  if (pCmdCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(cmdUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our characteristic");
  Serial.println(cmdUUID.toString().c_str());

// in loop()
...
      uint8_t array[4] = {03, 17, 01, 01};
      pCmdCharacteristic->writeValue(array, 4);
...
```
Try to figure out more.

xActTM, 2021-06-27:
Well, I checked under Open GoPro's python tutorials and it has [this graph](https://mermaid.ink/svg/eyJjb2RlIjoic2VxdWVuY2VEaWFncmFtXG5wYXJ0aWNpcGFudCBQQyBhcyBPcGVuIEdvUHJvIHVzZXIgZGV2aWNlXG5wYXJ0aWNpcGFudCBHb1Byb1xuR29Qcm8tLT4-UEM6IEFkdmVydGlzaW5nXG5Hb1Byby0tPj5QQzogQWR2ZXJ0aXNpbmdcbm5vdGUgb3ZlciBQQzogU2Nhbm5pbmdcblBDLT4-R29Qcm86IENvbm5lY3Rcbm5vdGUgb3ZlciBHb1BybywgUEM6IENvbm5lY3RlZFxuYWx0IElmIG5vdCBQcmV2aW91c2x5IFBhaXJlZFxuUEMgLT4-IEdvUHJvOiBQYWlyIFJlcXVlc3RcbkdvUHJvIC0-PiBQQzogUGFpciBSZXNwb25zZVxuZWxzZVxuJSUtXG5lbmRcbm5vdGUgb3ZlciBHb1BybywgUEM6IFBhaXJlZFxuUEMgLT4-IEdvUHJvOiBFbmFibGUgTm90aWZpY2F0aW9ucyBvbiBDaGFyYWN0ZXJpc3RpYyAxXG5QQyAtPj4gR29Qcm86IEVuYWJsZSBOb3RpZmljYXRpb25zIG9uIENoYXJhY3RlcmlzdGljIDJcblBDIC0-PiBHb1BybzogRW5hYmxlIE5vdGlmaWNhdGlvbnMgb24gQ2hhcmFjdGVyaXN0aWMgLi5cblBDIC0-PiBHb1BybzogRW5hYmxlIE5vdGlmaWNhdGlvbnMgb24gQ2hhcmFjdGVyaXN0aWMgTlxubm90ZSBvdmVyIEdvUHJvLCBQQzogUmVhZHkgdG8gQ29tbXVuaWNhdGUiLCJtZXJtYWlkIjp7InRoZW1lIjoiZGVmYXVsdCJ9fQ). It shows I have to pair after forming a connection, and I can't find anywhere in my code that indicates that it pairs, and I can't figure out how to pair after a connection.

chegewara, 2021-06-27:
With this i cant help, because i dont know how it performing pairing. With pin or not etc
You have more examples to play with security and pairing:
https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests/Arduino/security

xActTM, 2021-06-27:
I'm pretty sure the GoPro doesn't use a PIN to connect, or any kind of security. It just connects to it without any extra information, like on my laptop I can just click on it and it connects, it even shows that it connects on the GoPro.

xActTM, 2021-08-21:
Update: The GoPro uses some kind of BLE encryption, adding this line after initializing connects successfully.

```
BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
```

bjornjuul, 2022-01-30:
@guighub Sorry for the late comment, but I'm working on a similar project and trying to connect to a Gopro 9 via a Olimex ESP32 POE board.
Connection seems to go well until it finds the server. Then out of a sudden, the board hits a "Stack smashing protect failure!" and the board resets.

Would you mind sharing your final working code so I can try it out to see, where mine is going wrong?

chegewara, 2022-01-30:
> "Stack smashing protect failure!"

https://www.google.com/search?q=%22Stack+smashing+protect+failure!%22&oq=%22Stack+smashing+protect+failure!%22&aqs=chrome..69i57.561j0j1&sourceid=chrome&ie=UTF-8

guighub, 2022-01-31:
@bjornjuul Sure, I haven't tested this in a while and it seems to only work when it's plugged in to my computer but it should take some photos with different exposure values:
[GOPRO_BRACKET.txt](https://github.com/nkolban/esp32-snippets/files/7972182/GOPRO_BRACKET.txt)

TomRolander, 2022-02-14:
@bjornjuul were you able to get your code to work?  I'm having difficulty reading the Wifi AP SSID and Password.

bjornjuul, 2022-02-26:
@TomRolander H Tom. Sorry for the late reply, but was hit pretty hard by Covid, so haven't been able to work on the project.

Hope to back on my feet soon, and will certainly post updates when possible.

TomRolander, 2022-02-28:
@bjornjuul , so sorry to hear about your Covid ordeal!  I've been careful (and lucky;).  Stanford University has provided weekly covid testing for the past 6 months to staff which has been comforting.  Both because they've taken Covid very seriously, and that with frequent PCR testing I know whether I'm "positive".

The BLE code works perfectly without issues on the ESP32.  My problem has been getting the ESP32 talking to a GoPro HERO9 with WiFi.  I've actually given up on that approach to control the GoPro with WiFi and switched to using only BLE because my ESP32 is always in close range to the GoPro HERO9.  I will happily post my code which also includes a BLE client mode interface to a smartphone (iPhone or Android) BlueFruitConnect app.  I am curious if you've been able to get an ESP32 to control a GoPro after you've turned on the device WiFi.

alexrjohnston, 2023-04-08:
Hi Tom, would you be happy to share you code? BLE control with an arduino is exactly what I am after! Many thanks

alexrjohnston, 2023-04-08:
Hi @TomRolander , would you be happy to share yoru code? BLE control with an arduino is exactly what I am after! Many thanks

alexrjohnston, 2023-04-18:
@TomRolander would it be possible to share your code? tia!