nkolban/esp32-snippets #1085: Getting sings instead of value from characteristic

miag676, 2021-07-08:
Hello, I'm trying to read a value from my Inkbird BLE temperature sensor using an esp32. I'm using the BLEClient code, however, the problem is that I get the following output: "The characteristic value was: ⸮
⸮⸮". The code characteristic value is always a bit different, but it contains signs every time. If I look at the value on the BLE scanner app on my phone, everything works correctly. Does anyone know what could be causing this problem?

Thanks in advance!

chegewara, 2021-07-08:
The problem is that you are trying to print as string non printable values.
https://www.encyklopedialesna.pl/foto/hasla/0/0/ascii-001.jpg

miag676, 2021-07-08:
Thank you for your quick reply. I'm trying to get the a hexadecimal number(see the screenshot below). Do you maybe know a way to solve this, so that I could use the value further in the code?
[image]

chegewara, 2021-07-08:
You have to know protocol used by this sensor to interpret each value. Without it it is just a bunch of bytes without meaning.
Its like i would say it is now temperature of 22 in here, but you dont know if its K or C or F.

miag676, 2021-07-08:
Yes, I this case the first 4 bytes represent the temperature, however, how could I extract this hexadecimal number in my code. Currently this is how I get it: 
`    std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());`

But I'm not sure how to change the pRemoteCharacteristic to a string.

chegewara, 2021-07-08:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLECharacteristic.h#L64

miag676, 2021-07-08:
I'm new to programming in arduino, so I'm struggling with implementing the code you sent. I tired changing it to the following:
`    uint8_t* value = pRemoteCharacteristic->getData();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());`
But I am getting this error: 'class BLERemoteCharacteristic' has no member named 'getData'.
Could you provide me with a bit more information on how to use it?
Thanks for your help!

chegewara, 2021-07-08:
You have great tool you can use, it is `google`.
Try to search `arduino print bytes value`