nkolban/esp32-snippets #1091: Simple Server & Client for BLE with PIN/Password

JoFRT, 2021-09-14:
Hello,
After several evenings I can not suceed having a simple Server-Client couple sketches with a PIN/password security in Arduino.
I have 2 ESP32 DevKit1. And examples without security works well but I need to secure data.
Could you please help me ?

chegewara, 2021-09-14:
Do you have android smartphone?
Try secured example to work with nrf connect first to learn and understand how it works.

JoFRT, 2021-09-15:
Hello Chegewara,
Thanks for answering me.
Yes I have an Android smartphone
I think about a new test in the morning (french time). With nef connect I saw that finally it was possible to exchange with the server by refusing the PIN code request from the server, so not very secured finally.
I have to reply this test with the simple sketch StaticPIN.ino to ensure this issue is not due to my changes.
After the realization of my test I will give you the results.

JoFRT, 2021-09-16:
Hello Chegewara,
So I made a test with the combination of StaticPIN.ino and BLE_notify.ino:
      - to have security
      - to have always an adversing after a disconnection 
      - without Notification (security issue)
Everything is ok for sketch the ESP32 Server. (sometime I got this error "lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
", but maybe it is a side effect on connections/disconnections).

But impossible to have the corresponding sketch for the ESP32 client. I tried with BLE_client_passkey.ino but nothing happened, and with BLE_client_numeric_confirmation.ino or BLE_client_encrypted.ino I have a systematic reboot: below here is an extract of the serial monitor:
" _- Created client
 - Connected to server
remote BD_ADDR:address type = %d0pair status = success - Found our service
 - Found our characteristic
The characteristic value was: !
CORRUPT HEAP: Bad head at 0x3ffce134. Expected 0xabba1234 got 0x3ffce13c

abort() was called at PC 0x40084883 on core 1

Backtrace:0x400e1d69:0x3ffcdc300x40091f05:0x3ffcdc50 0x40097169:0x3ffcdc70 0x40084883:0x3ffcdcf0 0x40084a5d:0x3ffcdd20 0x401744c9:0x3ffcdd50 0x4016ce51:0x3ffce010 0x4016cdd9:0x3ffce060 0x40096ea3:0x3ffce090 0x4008367d:0x3ffce0b0 0x40097191:0x3ffce0d0 0x4016afed:0x3ffce0f0 0x400d1eb5:0x3ffce110 0x400d20d7:0x3ffce170 0x400d8f18:0x3ffce1d0 

ELF file SHA256: 0000000000000000

Rebooting..._
"

chegewara, 2021-09-16:
I am guessing you just copy/paste security client example, but it is probably not prepared to work with server on another esp32 without small adjustments.
It is bad and it is  good at the same time. Bad because you cant take 2 examples and flash just like that. Good, because you have to learn at least basics about BLE to use them.
For example in both examples UUIDs are different:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_client/BLE_client_passkey/BLE_client_passkey.ino#L9-L11
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_server/BLE_server_passkey/BLE_server_passkey.ino#L13-L14

From logs i dont know if you have connection and what crashed. You have to use tool to decode backtrace, for example:
https://github.com/me-no-dev/EspExceptionDecoder in arduino IDE.

JoFRT, 2021-09-17:
Hello Chegewara,
I have well applied same UUIDs but yes I have a lot to learn on BLE (and maybe on Cpp also).
I give you the results of decoder tool hoping that it means something for you.
I have made several tests this morning and in the backtrace result I can see some references on BLE_client_test.ino used just before (where there is an issue on a connection trial with a 2nd characteristic).
So I get this backtrace below with a modified sketch based on "BLE_server_passkey.ino".

here is the result copy of the monitor window:
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
The characteristic value was: ⸮
CORRUPT HEAP: Bad head at 0x3ffce224. Expected 0xabba1234 got 0x3ffce22c

abort() was called at PC 0x40084883 on core 1
"

backtrace:
" Decoding stack results
0x400e21ad: nvs::NVSPartitionManager::open_handle(char const*, char const*, nvs_open_mode_t, nvs::NVSHandleSimple**) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/intrusive_list.h line 120
0x40091f05: rtc_clk_cpu_freq_set_config at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32/rtc_clk.c line 655
0x40097169: spi_flash_hal_configure_host_io_mode at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32/include/hal/spi_flash_ll.h line 275
0x40084a5d: lock_release_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 181
0x40174909: std::money_get    > >::_M_extract (std::istreambuf_iterator   >, std::istreambuf_iterator   >, std::ios_base&, std::_Ios_Iostate&, std::string&) const at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/no-rtti/libstdc++-v3/include/bits/locale_facets_nonio.tcc line 178
0x4016d291: rc_cal at phy_chip_v7_cal.c line 1063
0x4016d219: rc_cal at phy_chip_v7_cal.c line 1046
0x40096ea3: soc_hal_unstall_core at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32/include/hal/soc_ll.h line 43
0x4008367d: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 48
0x40097191: spi_flash_hal_common_command at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/spi_flash_hal_common.inc line 153
0x4016b42d: phy_get_romfunc_addr at phy_chip_v7.c line 3119
0x400d1dc5: xxLcdShiftPrint(String) at C:\Users\xxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc2\cores\esp32/WString.h line 344
0x400d2077: connectToServer() ([MY_LOCATION]\2_Prog\BLE_client_test/BLE_client_test.ino line 335
0x400d935c: btStart at C:\Users\xxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc2\cores\esp32\esp32-hal-bt.c line 44
"

chegewara, 2021-09-17:
From logs i can tell it is not related to BLE. You seem to have memory corruption issues (maybe you are passing empty value as string).
```
0x400d1dc5: xxLcdShiftPrint(String) at C:\Users\xxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc2\cores\esp32/WString.h line 344
```
You need to investigate where you call this function and what you pass to it.

JoFRT, 2021-09-17:
Thanks for your very quick answer I still do not why the backtrace corresponded to the old download sketch. Your analysis will help me for the other sketch. 
I have replayed this evening and this time I have the good backtrace, I give it to you, I tried to see if the error comes from my changes, but I do not think so.
"Decoding stack results
0x400e21ad: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x40091f05: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 129
0x40097169: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x40084883: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 139
0x40084a5d: __retarget_lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 323
0x40174909: _vfiprintf_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c line 853
0x4016d291: fiprintf at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c line 48
0x4016d219: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c line 58
0x40096ea3: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 245
0x4008367d: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 305
0x40097191: free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c line 46
0x4016b42d: operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc line 49
0x400d1dc5: connectToServer(BLEAddress) at c:\users\xxxxx\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\include\c++\8.4.0\ext/new_allocator.h line 116
0x400d2077: loop() ([MY_LOCATION]\Elec\8 - µController\ESP32_VROOM3 at Arduino)\2_Prog\BLE_client_passkey/BLE_client_passkey.ino line 173
0x400d935c: loopTask(void*) at C:\Users\xxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc2\cores\esp32\main.cpp line 46
"

chegewara, 2021-09-17:
Seems you are trying to delete/free something that already has been deleted or free.

JoFRT, 2021-09-17:
Hummm hum... 
I rather first to focus on simple sketches. For that I tried client & server with BLE passkey sketches. I have aligned UUIDs and replace ESP_LOG by serial.println in order to get what's happening. 
The client see the server but failed in the first "if" condition of the onResult function of the MyAdvertisedDeviceCallbacks class. I can not understand why it failed there.
Is it normal? is there simple server client sketches working together to be familiar with BLE security? 
Thanks in advance for your help

chegewara, 2021-09-18:
I dont know why it does fail in first `if`.
The simplest start with security is StaticPIN example. It is setup as is and pin is `123456` and any security example should do (or maybe passkey example). You should start with simple server and client and try to find when they work together.

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_server/BLE_server.ino#L32
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L80

I am guessing it is not working, because the name is too long:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_server/BLE_server.ino#L20

try 5 or less name long.

JoFRT, 2021-09-18:
Many thanks for your answer and your time.
The name is defined in 5 length.
My sketches server client without encrytion and PIN work well: read and write operation.
I have made another test this morning, and I  can see that the UUID is not seen by the client.
But it csn see UUID of an iTag (without encryption and PIN).
So I change the condition on UUID check by Address check. In this case the client can read the value sent by the server.
But just after I have the same kind of error of the last one I have exchanged.
I will make another test this evening or tomorrow to ensure that Arduino compiles with the good and last library.
Thanks again.

chegewara, 2021-09-18:
> I have made another test this morning, and I can see that the UUID is not seen by the client.

In that case check if you can see service UUID in scan mode with nrf connect.

JoFRT, 2021-09-19:
Hello,

In scan mode (Flags & Services), I can not read the UUID. Flags are 0x06 (for the iTag it is 0x05 and I can get the UUID service as an example)

After a connection trial, the smartphone can get UUID (Service and characteristic) but failed on connection because of Passkey but I put the nrf Connect log below just in case.

The next step I am thinking about is to uninstall all Arduino and librairies and install them again by taking only librairies of "esp32-snippets-master". But I am not ready for that for the moment because I am afraid to spend time again for other skeches not related to ESP32.
Maybe I can try to establish a client for StaticPIN.ino before.

"nRF Connect, 2021-09-19
ESP32 (7C:62:BD:4C:34:9E)
V	06:39:53.431	Connecting to 7C:62:BD:4C:34:9E...
D	06:39:53.431	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	06:39:53.697	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D	06:39:53.708	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	06:39:53.708	Connected to 7C:62:BD:4C:34:9E
D	06:39:53.731	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	06:39:53.748	[Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_CONSENT (3)
I	06:39:54.173	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
I	06:39:54.429	Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
D	06:39:56.344	[Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_PIN (0)
D	06:40:00.146	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_NONE (10)
I	06:40:00.146	Bonding failed
V	06:40:00.168	Discovering services...
D	06:40:00.168	gatt.discoverServices()
D	06:40:00.176	[Callback] Services discovered with status: 0
I	06:40:00.176	Services discovered
V	06:40:00.177	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Central Address Resolution [R] (0x2AA6)
Unknown Service (4fafc201-1fb5-459e-8fcc-c5c9c331914b)
- Unknown Characteristic [R W] (beb5483e-36e1-4688-b7f5-ea07361b26a8)
D	06:40:00.177	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	06:40:03.166	[Callback] Connection state changed with status: 22 and new state: DISCONNECTED (0)
E	06:40:03.166	Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST
I	06:40:03.166	Disconnected
D	06:40:03.216	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
"

chegewara, 2021-09-19:
Now you know the reason. Try this:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_server/BLE_server.ino#L35

JoFRT, 2021-09-20:
Hello Chegewara,
It works better with that.
Many many thanks for your help and your time.

closed 2021-10-10 by JoFRT