nkolban/esp32-snippets #1092: advertisedDevice.getAppearance() always 0

FlorianBogeschdorfer, 2021-09-19:
Awesome lib! Thank you. I can get RSSI and after a hotfix also the name. However, the appearance is always 0. Even with my speakers which identify by name as "BL Xtreme 2" (Pioneer). Am i misunderstanding something about appearance?

Thank you

chegewara, 2021-09-27:
There is 2 appearance options:
1) in advertising, then for example HID device can display proper icon,
2) in characteristic: Service 0x1800 with characteristic 0x2A01

antonelliCP, 2023-04-27:
I am seeing this problem as well with ESP32
I am able to add a record to the advertising packet for appearance (with BLEadvertising->setappearance(int16_t x)
but x never appears in nRFconnect, or light blue charactersitic 0x2A01) as anything but 00. 

I don't see any more input on this open issue, so Im wondering where this stands.