nkolban/esp32-snippets #1094: Is it possible to BLE client(ESP32) connect to multi BLE server(ESP32)?

sssuun, 2021-09-23:
Hi, I tried to use one ESP32 as a client and one as a server and get RSSI from the server to monitor the client. It works completely.

but now I don't know how to make the client connect to multi esp32 ble server (1:m) and get RSSI from every server for calculate the location.

My question is. Can client connect to multiple server? if it can how many limit of server for connect? Can client get RSSI from every server?

Here my code for 1:1 communication
[BLE_client.txt](https://github.com/nkolban/esp32-snippets/files/7220070/BLE_client.txt)

BLE_server is an example in library.

Sorry for my bad English.

sssuun, 2021-09-25:
I can connected multiple server by following this. #915 

but I have new issue about RSSI value not keep looping on serial monitor.
here's my code.
[BLE_ClientV2.txt](https://github.com/nkolban/esp32-snippets/files/7230330/BLE_ClientV2.txt)

sssuun, 2021-09-26:
here's serial monitor showing.

> Starting Arduino BLE Client application...
Potential Servers: Name: ESP32 Server A, Address: 24:6f:28:9e:a5:c2, serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd, txPower: 3
RSSI A : -40
Potential Servers: Name: ESP32 Server A, Address: 24:6f:28:9e:a5:c2, serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd, txPower: 3
RSSI B : -41
Potential Servers: Name: ESP32 Server B, Address: 24:6f:28:9d:74:56, serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623, txPower: 3
RSSI B : -47
Forming a connection to Server A with MAC Address: 24:6f:28:9e:a5:c2
Server A Service found
Server A Characterstics found
We are now connected to Server A
RSSI A :-46
Forming a connection to Server B with MAC Address: 24:6f:28:9d:74:56
Server B Service found
Server B Characterstics found
We are now connected to Server B
RSSI B :-50
>

sssuun, 2021-09-26:
### Debug Messages:
```
Starting Arduino BLE Client application...
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -49
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: cdfad86aae378e80fa403f45aff5dd7c
[D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4553503332205365727665722041
[D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: ESP32 Server A
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 03
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 3
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
Potential Servers: Name: ESP32 Server A, Address: 24:6f:28:9e:a5:c2, serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd, txPower: 3
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -48
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: 2356e142a5b052889c46b77d27c1e269
[D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4553503332205365727665722042
[D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: ESP32 Server B
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 03
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 3
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
Potential Servers: Name: ESP32 Server B, Address: 24:6f:28:9d:74:56, serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623, txPower: 3
Forming a connection to Server A with MAC Address: 24:6f:28:9e:a5:c2
[I][BLEDevice.cpp:614] addPeerDevice(): add conn_id: 0, GATT role: client
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
Server A Service found
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: 3383a686-a53c-42e4-a88c-e0e1de4c4bda
[E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
Server A Characterstics found
We are now connected to Server A
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI A :-50
Forming a connection to Server B with MAC Address: 24:6f:28:9d:74:56
[I][BLEDevice.cpp:614] addPeerDevice(): add conn_id: 1, GATT role: client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[... cut ...]
[D][BLEClient.cpp:177] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
Server B Service found
[D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: 2916187f-e651-4c42-9144-d91262892b22
[E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
Server B Characterstics found
We are now connected to Server B
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI B :-47
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!

```

chegewara, 2021-09-26:
Everything seems to work fine. Looks like code stuck somewhere, but hard to say where.

My advise is to find in code where it takes semaphore for `portMAX_DELAY` and change it to 100 dor example. It should be done long time ago, but as you can see library is not updated for long time and not becuase it is perfect.

sssuun, 2021-09-26:
Hi @chegewara thank you for your reply.

but can you explain more about `portMAX_DELAY` ?
I need to change the code of `portMAX_DELAY` in this file to 1000?
(https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/FreeRTOS.cpp#L68)
or I should try something about freeRTOS in BLE_client?

sorry for wasting your time. I just new in ESP32

chegewara, 2021-09-26:
Yes, it is exactly the file from link.
This semaphore is sometimes just blocking code forever. Like i said, it should be changed long time ago, but there is no one who would like to do it and make PR.

sssuun, 2021-09-27:
I'm take your advice and its work! but have another problem about my RSSI's server_A is the same. but server_B can get new RSSI.
```
Starting Arduino BLE Client application...
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -48
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: 2356e142a5b052889c46b77d27c1e269
[D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4553503332205365727665722042
[D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: ESP32 Server B
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 03
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 3
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
Potential Servers: Name: ESP32 Server B, Address: 24:6f:28:9d:74:56, serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623, txPower: 3
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -37
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: cdfad86aae378e80fa403f45aff5dd7c
[D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4553503332205365727665722041
[D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: ESP32 Server A
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 03
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 3
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
Name: ESP32 Server A, Address: 24:6f:28:9d:74:e6, serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd, txPower: 3
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -63
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 0c
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0xff (), length: 10, data: 4c0010060d19f7946108
[D][BLEAdvertisedDevice.cpp:449] setManufacturerData(): - manufacturer data: 4c0010060d19f7946108
Name: , Address: 5e:a1:d6:1d:05:cb, manufacturer data: 4c0010060d19f7946108, txPower: 12
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -38
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: cdfad86aae378e80fa403f45aff5dd7c
[D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4553503332205365727665722041
[D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: ESP32 Server A
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 03
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 3
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
Name: ESP32 Server A, Address: 24:6f:28:9d:74:e6, serviceUUID: 7cddf5af-453f-40fa-808e-37ae6ad8facd, txPower: 3
[D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -47
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: 2356e142a5b052889c46b77d27c1e269
[D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 69e2c127-7db7-469c-8852-b0a542e15623
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x12 (), length: 4, data: 12004000
[D][BLEAdvertisedDevice.cpp:388] parseAdvertisement(): Unhandled type: adType: 18 - 0x12
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 14, data: 4553503332205365727665722042
[D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: ESP32 Server B
[D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x0a (), length: 1, data: 03
[D][BLEAdvertisedDevice.cpp:530] setTXPower(): - txPower: 3
[... cut ...]
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI A :-37
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI B :-50
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI A :-37
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI B :-50
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI A :-37
[D][BLEClient.cpp:492] handleGAPEvent(): BLEClient ... handling GAP event!
RSSI B :-51

```

chegewara, 2021-09-27:
Whit this i cant help you, sorry, i just dont know what may be the issue.

sssuun, 2021-09-30:
thank you for reply @chegewara .

I try to add one more server is Server C and its the same.
I think this happen because when I try to use getRSSI() the first and second server that connect it will use getRSSI() until the new one connected and when the last server connected it can use getRSSI() over and over again.

my problem is I want to set client can use getRSSI() again and again to all connected server.

sssuun, 2021-10-05:
I'm can't find anything to fix this. 

and now I think I change the Client to the Scan for scanning server and get RSSI.
but when I scan I got too many device. It have anything to get the scan only specific device? (UUID or MAC)

chegewara, 2021-10-05:
@sssuun are you sure those devices really advertise with UUIDs? I am only guessing those are other esp32 devices. Best way to confirm it is to use nrf connect (android version).

sssuun, 2021-10-06:
I use nrf connect in iOS and it show only service UUID.

[image]

sssuun, 2021-10-06:
I should try this right?
```cpp
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {

       Serial.println(advertisedDevice.getServiceUUID().toString().c_str() == "7cddf5af-453f-40fa-808e-37ae6ad8facd")
      
      /*if (advertisedDevice.getServiceUUID().toString().c_str() == "7cddf5af-453f-40fa-808e-37ae6ad8facd"){
        int rssi_a = advertisedDevice.getRSSI();
        Serial.printf("Name: %s // Address: %s // RSSI: %d \n\n", advertisedDevice.getName(), 
        advertisedDevice.getAddress().toString().c_str(), rssi_a);
      }

      if (advertisedDevice.getServiceUUID().toString().c_str() == "69e2c127-7db7-469c-8852-b0a542e15623"){
        int rssi_b = advertisedDevice.getRSSI();
        Serial.printf("Name: %s // Address: %s // RSSI: %d \n\n", advertisedDevice.getName(), 
        advertisedDevice.getAddress().toString().c_str(), rssi_b); 
      }

      if (advertisedDevice.getServiceUUID().toString().c_str() == "407fc20d-d5f5-4176-bb4d-cc9c1968f9f6"){
        int rssi_c = advertisedDevice.getRSSI();
        Serial.printf("Name: %s // Address: %s // RSSI: %d \n\n", advertisedDevice.getName(), 
        advertisedDevice.getAddress().toString().c_str(), rssi_c);
      }*/
    }
};
```

chegewara, 2021-10-06:
Yes, something like that.

sssuun, 2021-10-06:
Thank you for advice and idea @chegewara 
Now I can filter but using by name and I can print RSSI and Its work! 
but I want to calculate something by using rssi_a and rssi_b in `void loop();` but get this error.

>BLE_scan_test:52:16: error: 'rssi_a' was not declared in this scope
   Serial.print(rssi_a);
                ^
BLE_scan_test:54:18: error: 'rssi_b' was not declared in this scope
   Serial.println(rssi_b);
                  ^
Multiple libraries were found for "BLEDevice.h" 
>

### Code
```cpp
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

static BLEUUID serviceUUID_A("7cddf5af-453f-40fa-808e-37ae6ad8facd");

static BLEUUID serviceUUID_B("69e2c127-7db7-469c-8852-b0a542e15623");

static BLEUUID serviceUUID_C("407fc20d-d5f5-4176-bb4d-cc9c1968f9f6");

int scanTime = 0; //In seconds

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {

      if (advertisedDevice.getName() == "ESP32 Server A"){
        int rssi_a = advertisedDevice.getRSSI();
       /* Serial.printf("Name: %s // Address: %s // RSSI A: %d \n\n", advertisedDevice.getName(), 
        advertisedDevice.getAddress().toString().c_str(), rssi_a);*/
      }

      if (advertisedDevice.getName() == "ESP32 Server B"){
        int rssi_b = advertisedDevice.getRSSI();
       /* Serial.printf("Name: %s // Address: %s // RSSI B: %d \n\n", advertisedDevice.getName(), 
        advertisedDevice.getAddress().toString().c_str(), rssi_b);*/
      }
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
}

void loop() {
  Serial.print(rssi_a);
  delay(1000);
  Serial.println(rssi_b);
  delay(1000);
}
```

sssuun, 2021-10-06:
I have a question. why it scan forever but it isn't in `void loop();` ? and it possible to use local variable in loop?

I try this in loop because if I use only `advertisedDevice.getRSSI();` I will doesn't know what RSSI of device I get and error again.
```cpp
void loop() {
     if (advertisedDevice.getName() == "ESP32 Server A"){
        int rssi_a = advertisedDevice.getRSSI();
        printf("RSSI A: %d \n\n", rssi_a);
        }
  delay(1000);
}
```
### Error

> BLE_scan_test:53:10: error: 'advertisedDevice' was not declared in this scope
      if (advertisedDevice.getName() == "ESP32 Server A"){
          ^
Multiple libraries were found for "BLEDevice.h"
>

chegewara, 2021-10-06:
again, this time `advertisedDevice` is local variable withing `void onResult(BLEAdvertisedDevice advertisedDevice) {`
You can make `int rssi_a` global, but you have whole google to learn such basic stuff.

sssuun, 2021-10-06:
thank for advice again @chegewara I try to look at BLE_client and get something like you said and It works.
but sometime get this error on monitor.
> abort() was called at PC 0x40174fd3 on core 0
>
### Code
```cpp
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

static BLEAdvertisedDevice* Scan_A;
static BLEAdvertisedDevice* Scan_B;

static bool Found_A = false;
static bool Found_B = false;

int scanTime = 0; //In seconds

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {

      if (advertisedDevice.getName() == "ESP32 Server A"){
        Scan_A = new BLEAdvertisedDevice(advertisedDevice);
        Found_A = true;
      }
      else{
        Found_A = false;
      }

      if (advertisedDevice.getName() == "ESP32 Server B"){
        Scan_B = new BLEAdvertisedDevice(advertisedDevice);
        Found_B = true;
      }
      else{
        Found_B = false;
      }
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());  
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
}

void loop() {
    if(Found_A){
      int rssi_a = Scan_A->getRSSI();
      Serial.printf("RSSI A: %d \n\n", rssi_a);
      delay(1000);
    }
  

    if(Found_B){
      int rssi_b = Scan_B->getRSSI();
      Serial.printf("RSSI B: %d \n\n", rssi_b);
      delay(1000);
    }
  }
```

chegewara, 2021-10-07:
```
void loop() {
    if(Found_A){
      int rssi_a = Scan_A->getRSSI();
      Serial.printf("RSSI A: %d \n\n", rssi_a);
      delete Scan_A;
      Found_A = false;
      delay(1000);
    }
```
Sure, you can try NimBLE if you want to.

sssuun, 2021-10-07:
I try your code. but when I add new server (server c) It get error again.
or esp32 not support to do this thing.

chegewara, 2021-10-07:
https://github.com/espressif/arduino-esp32#decoding-exceptions

sssuun, 2021-10-07:
### Exception Decoder
```
Decoding stack results
0x4008ec74: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008eef1: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x40175227: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x4017526e: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x40174b95: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc line 268
0x40174c70: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400d16ef: MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice) at C:\Users\Sun\Desktop\Project\BLE\BLE_scan_test/BLE_scan_test.ino line 30
0x400d4311: BLEScan::handleGAPEvent(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\Sun\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEScan.cpp line 125
0x400d322a: BLEDevice::gapEventHandler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\Sun\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEDevice.cpp line 270
0x400f0ecd: btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c line 54
0x400ea95e: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c line 163
0x4008ff66: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

https://github.com/sssuun/ESP32_BLEScanFilter/blob/ce066e23bd9f792a8e538da53b5d4a3bd589fbaa/BLE_scan_test.ino#L1-L91

sssuun, 2021-10-07:
If I want to add semaphore what should I do or what should I study about ?

sssuun, 2021-10-07:
I try your code and it the same.
```
Decoding stack results
0x4008ec74: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008eef1: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x4017529b: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x401752e2: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x40174c09: __cxxabiv1::__cxa_allocate_exception(std::size_t) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc line 268
0x40174ce4: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400d16f7: MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice) at C:\Users\Sun\Desktop\Project\BLE\BLE_scan_test/BLE_scan_test.ino line 30
0x400d4381: BLEScan::handleGAPEvent(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\Sun\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEScan.cpp line 125
0x400d329a: BLEDevice::gapEventHandler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at C:\Users\Sun\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEDevice.cpp line 270
0x400f0f3d: btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c line 54
0x400ea9ce: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c line 163
0x4008ff66: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

```cpp
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {

      if (advertisedDevice.isAdvertisingService(serviceUUID_A)){
        Scan_A = new BLEAdvertisedDevice(advertisedDevice);
        Found_A = true;
      }
      else{
        Found_A = false;
      }
     if(Scan_A == NULL){
        Scan_A = new BLEAdvertisedDevice(advertisedDevice);
        Found_A = true;
      }

      if (advertisedDevice.isAdvertisingService(serviceUUID_B)){
        Scan_B = new BLEAdvertisedDevice(advertisedDevice);
        Found_B = true;
      }
      else{
        Found_B = false;
      }
      if(Scan_B == NULL){
        Scan_B = new BLEAdvertisedDevice(advertisedDevice);
        Found_B = true;
      }

      if (advertisedDevice.isAdvertisingService(serviceUUID_C)){
        Scan_C = new BLEAdvertisedDevice(advertisedDevice);
        Found_C = true;
      }
      else{
        Found_C = false;
      }
      if(Scan_C == NULL){
        Scan_C = new BLEAdvertisedDevice(advertisedDevice);
        Found_C = true;
      }
    }
};
```
```cpp
void loop() {
    if(Found_A){
      int Rssi_A = Scan_A->getRSSI();
      Serial.printf("RSSI A: %d \n\n", Rssi_A);
      delete Scan_A;
      Scan_A = NULL;
      Found_A = false;
      delay(500);
    }

    if(Found_B){
      int Rssi_B = Scan_B->getRSSI();
      Serial.printf("RSSI B: %d \n\n", Rssi_B);
      delete Scan_B;
      Scan_B = NULL;
      Found_B = false;
      delay(500);
    }

    if(Found_C){
      int Rssi_C = Scan_C->getRSSI();
      Serial.printf("RSSI C: %d \n\n", Rssi_C);
      delete Scan_C;
      Scan_C = NULL;
      Found_C = false;
      delay(500);
    }
  }
```

sssuun, 2021-10-07:
Thank you so much @chegewara it work well. 
### Monitor
> RSSI A: -56 	RSSI B: -60 	RSSI C: -50 	RSSI D: -50 	
RSSI A: -57 	RSSI B: -50 	RSSI C: -50 	RSSI D: -51 	
RSSI A: -58 	RSSI B: -50 	RSSI C: -50 	RSSI D: -50 	
RSSI A: -56 	RSSI B: -60 	RSSI C: -50 	RSSI D: -50 	
>

### Code
https://github.com/sssuun/ESP32_BLEScanFilter/blob/main/BLE_scan_test.ino

In future I will try about increase the server signal strength for the accurate on position.

ercntrn28, 2022-05-31:
Hi,
I am making Indoor positioning system project and I am using 3 Qty ESP8266 device. I tried your code but It is not working. Did you make anything change? Please Can you help me about this topic?

Thanks.

ercntrn28, 2022-05-31:
Please Can you help me? I have ESP32 WROOM32-U device. Also I select to device like below picture. Also I am using 1.0.2 version of ESP32. I tried with lst version(1.0.6) but It didn't work

ercntrn28, 2022-05-31:
Code: 
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

//UUIDs of what server we want to filtering.
static BLEUUID serviceUUID_A("0000fdee-0000-1000-8000-00805f9b34fb");
static BLEUUID serviceUUID_B("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
static BLEUUID serviceUUID_C("407fc20d-d5f5-4176-bb4d-cc9c1968f9f6");
static BLEUUID serviceUUID_D("14fb4533-1997-4bbe-a567-554d2da89738");

//Set "Scan_x" for condition statement.
static BLEAdvertisedDevice* Scan_A ;
static BLEAdvertisedDevice* Scan_B ;
static BLEAdvertisedDevice* Scan_C ;
static BLEAdvertisedDevice* Scan_D ;

//Set "Found_x" for condition statement.
static bool Found_A = false;
static bool Found_B = false;
static bool Found_C = false;
static bool Found_D = false;

//Set scanTime to forever "scanTime = 0".
int scanTime = 2; //In seconds

//Function when found device and can use in void loop(); for do other things.
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      if (advertisedDevice.isAdvertisingService(serviceUUID_A)) {

        if (Scan_A == NULL) {
          Found_A = true;
          Scan_A = new BLEAdvertisedDevice(advertisedDevice);
        }
      }

      if (advertisedDevice.isAdvertisingService(serviceUUID_B)) {
        if (Scan_B == NULL) {
          Found_B = true;
          Scan_B = new BLEAdvertisedDevice(advertisedDevice);
        }
      }

    }

};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->start(scanTime);
}

void loop() {
  if (Found_A) {
    int Rssi_A = Scan_A->getRSSI();
    Serial.printf("RSSI A: %d \t", Rssi_A);
    delete Scan_A;
    Scan_A = NULL;
    Found_A = false;
    delay(500);
  }

  if (Found_B) {
    int Rssi_B = Scan_B->getRSSI();
    Serial.printf("RSSI B: %d \t", Rssi_B);
    delete Scan_B;
    Scan_B = NULL;
    Found_B = false;
    delay(500);
  }
  Serial.print("\n");
  delay(1000);
}