nkolban/esp32-snippets #1099: ESP32 BLE Arduino Version 2.0.0 in Platformio

marcopiai, 2021-11-02:
Dear
I'm using ESP32 BLE Arduino library on arduino ide and works fine. (I'm using setStaticPin and disconnect funtion).
I'm trying to port the project in Platformio and the library version available in Platformio is the older 1.0.1.
This version not support the function setStaticPin and disconnect.
I have tryed to copy the 2.0.0 library from arduino filder (AppData/Arduino15 etc) but not works, it generate some error on RTOS library included.
 any suggest ?

Many thanks

marcopiai, 2021-11-02:
I have found half solution, importing project from arduino i can select use "Arduino library" and this add to platformio.ini lib_extra_dirs = ~/Documents/Arduino/libraries
but not the BLE library in appData/etc/etc/etc

CLDiego, 2021-11-02:
Rather than trying to use the arduino libraries directly. Try to understand how Platformio works. The repo containing the [Arduino Core](https://github.com/espressif/arduino-esp32) incorporates this library (albeit with some changes). 

When working with PlatformIO choosing the arduino framework will load the above repo.

Hope that helps

chegewara, 2021-11-02:
Just like @CLDiego said, dont use this or any other repo with pio. PIO is using arduino version of BLE, which is clone of this repo with few fixes/updates.

marcopiai, 2021-11-03:
> 
> 
> Rather than trying to use the arduino libraries directly. Try to understand how Platformio works. The repo containing the [Arduino Core](https://github.com/espressif/arduino-esp32) incorporates this library (albeit with some changes).
> 
> When working with PlatformIO choosing the arduino framework will load the above repo, so you can call the BLE library as usual.
> 
> Hope that helps.
Many thanks for reply.
I admit that platformio is a little more complicated to use.
I'm trying to understand platformio, but is not simple.
I had understood that using arduino framework it would have to work without special configurations with the libraries already installed in arduino ide, but for some reason it didn't work and I messed up
now i will try with a new clean project i hope to solve.

Marco

marcopiai, 2021-11-04:
> 
> 
> Just like @CLDiego said, dont use this or any other repo with pio. PIO is using arduino version of BLE, which is clone of this repo with few fixes/updates.

closed 2021-11-04 by marcopiai

bilogic, 2024-11-02:
> Just like @CLDiego said, dont use this or any other repo with pio. PIO is using arduino version of BLE, which is clone of this repo with few fixes/updates.

where is the repository of the arduino version of BLE?

chegewara, 2024-11-04:
https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE

bilogic, 2024-11-04:
Thanks, the versioning is really confusing. I mean there is a `version`, then there is git's `tag` and they aren't indicated clearly in PlatformIO.

Package managers should read version off the tag, e.g. PHP's composer.. just saying