nkolban/esp32-snippets #1100: error: 'ringbuf_type_t' has not been declared

PuneetDadhich, 2021-11-10:
```
Arduino\libraries\ESP32_BLE_Arduino\src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
  Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
                            ^~~~~~~~~~~~~~
Multiple libraries were found for "BLEDevice.h"
```

chegewara, 2021-11-10:
Did you try to search for this in other issues?

Haydnoleson, 2021-11-16:
I had the same issue, then I deleted the ESP32_BLE_ARDUINO folder in the libraries and problem solved. The necessary BLEDevice library is included in the arduino libraries already.

vishnumaiea, 2021-12-12:
> I had the same issue, then I deleted the ESP32_BLE_ARDUINO folder in the libraries and problem solved. The necessary BLEDevice library is included in the arduino libraries already.

I was getting same error with DFRobot Firebeetle-ESP32-E board. Deleted the ESP32_BLE_ARDUINO library and the code got compiled. Thanks :)

closed 2021-12-12 by chegewara

geofrancis, 2022-07-28:
can you post a working binary, its flashing but im getting flash errors so can you post a working binary so i can get the rest of the hardware tested and rule out a hardware issue.

orion253, 2023-05-22:
> I had the same issue, then I deleted the ESP32_BLE_ARDUINO folder in the libraries and problem solved. The necessary BLEDevice library is included in the arduino libraries already.

Where did you delete it from? When I delete this folder in my Documents/Arduino/libraries , I get the following error:

Compilation error: BLEDevice.h: No such file or directory

Anyone can help? I need to finish a project for uni and this is a key step :(

vishnumaiea, 2023-05-23:
@orion253 If there are conflicting libraries with the same header file names, we need to delete the unwanted ones from the "libraries" folder in order to force the Arduino IDE to use the correct library. Arduino IDE gives no control over the library name, version, or author. If you are using ESP32 boards from DFRobot, then you need to delete any other BLE libraries from the "libraries" folder. That is because the DFRobot board installation comes with its own BLE libraries that are not fully compatible with other ones.

If you are getting a BLEDevice.h missing error, that means the compiler could not find a suitable BLE library anywhere. Try to install a compatible BLE library for the board you are using.

orion253, 2023-05-23:
> @orion253 If there are conflicting libraries with the same header file names, we need to delete the unwanted ones from the "libraries" folder in order to force the Arduino IDE to use the correct [...]

That helps a lot, thank you @vishnumaiea ! 
By any chance, have you come across a tutorial online to learn BLE on Arduino IDE? I'm trying to convert the data from integers to byte arrays before sending it to the phone, and can't seem to find online resources that don't rely on the espressif IDF ?

Thanks!

JebilonRix, 2023-07-24:
> It is worked.

cwoodhouse, 2023-10-08:
I could never get it to work using an M5-Stack Atom Lite.

Mr-KID-github, 2024-02-18:
[image]
First , you need to delect the ESP32_BLE_Arduino library,
then, you need to select  suitable version,
in face , Version 2.0.11 of esp32 is available

JulianneJager, 2024-03-14:
I have been agonizing over this issue for hours:

> In file included from c:\Users\julia.DESKTOP-TSHJUVK\.arduinoIDE\libraries\ESP32_BLE_Arduino\src/BLEAdvertising.h:15,
>                  from c:\Users\julia.DESKTOP-TSHJUVK\.arduinoIDE\libraries\ESP32_BLE_Arduino\src/BLEServer.h:19,
>                  from c:\Users\julia.DESKTOP-TSHJUVK\.arduinoIDE\libraries\ESP32_BLE_Arduino\src/BLEDevice.h:18,
>                  from C:\Users\julia.DESKTOP-TSHJUVK\AppData\Local\Temp\.arduinoIDE-unsaved2024214-29068-1g44lkb.7pik\BLE_notify\BLE_notify.ino:22:
> c:\Users\julia.DESKTOP-TSHJUVK\.arduinoIDE\libraries\ESP32_BLE_Arduino\src/FreeRTOS.h:61:28: error: 'ringbuf_type_t' has not been declared
>   Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);
>                             ^~~~~~~~~~~~~~
> 
> Using library ESP32 BLE Arduino at version 1.0.1 in folder: C:\Users\julia.DESKTOP-TSHJUVK\.arduinoIDE\libraries\ESP32_BLE_Arduino 
> exit status 1
> 
> Compilation error: exit status 1

From what I see, others have solved it by deleting ESP32_BLE_Arduino and reinstalling it, but that hasn't changed anything for me. I've even tried deleting Arduino IDE and reinstalling an older version. That didn't work either. I tried deleting the FreeRTOS package from the library, but caused issues with dependencies so I reinstalled it again. Does anyone have other suggestions? 

I want to connect my ESP32 WROOM to an apple phone. Is there another library that I could use instead?

StultusRex, 2024-03-18:
> I have been agonizing over this issue for hours:
>
>
> From what I see, others have solved it by deleting ESP32_BLE_Arduino and reinstalling it, but that hasn't changed anything for me. I've even tried [...]

had the same issue, what I did to make it work were: 
1st - I re-installed my ESP32 board, through the boards manager.
2nd - uninstall/delete the ESP32_BLE_Arduino library and also the ArduinoBLE library. For the ArduinoBLE package, I just moved it outside my library folder just incase I'll be needing it in the future.

ManKhunt, 2024-07-03:
> had the same issue, what I did to make it work were: 1st - I re-installed my ESP32 board, through the boards manager. 2nd - uninstall/delete the ESP32_BLE_Arduino library and also the ArduinoBLE [...]

Hi StultusRex,

as you mentioned your ans. Like that i done in my system/Arduino IDE, but after doing all things i got "BLEDevice.h"  error.
Error :- "BLESDevice.h" No such directory.

for this if you have any solution then let suggest and I'll work on it, try to resolve my errors.
thank you

strgPeter, 2024-11-01:
After deleting the ESP32_BLE_ARDUINO folder I get the following error:

C:\Users\peter\Documents\Arduino\Bluetooth\test-1\test-1.ino:6:10: fatal error: BLEDevice.h: No such file or directory
    6 | #include <BLEDevice.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: BLEDevice.h: No such file or directory

Can someone tell me what im doing wrong?

Dharsh511, 2024-11-21:
> I have been agonizing over this issue for hours:
>
>
> From what I see, others have solved it by deleting ESP32_BLE_Arduino and reinstalling it, but that hasn't changed anything for me. I've even tried [...]

i get the same error now? how did you rectify it?

Dharsh511, 2024-11-21:
> Hi @StultusRex
>
> as you mentioned your ans. Like that i done in my system/Arduino IDE, but after doing all things i got "BLEDevice.h" error. Error :- "BLESDevice.h" No such directory.
>
> for this if [...]

how did you resolve it? i have the same error now

Philamagron, 2025-04-29:
> I had the same issue, then I deleted the ESP32_BLE_ARDUINO folder in the libraries and problem solved. The necessary BLEDevice library is included in the arduino libraries already.

Thanks!
In my IDE (2.3.6), the library was not installed (maybe because i frist installed it with the old version of arduino IDE 1.8.13), but the folder was still there. 
I suppressed the ESP32_BLE_ARDUINO manually, and then everything was fine.