nkolban/esp32-snippets #1104: Bad BLE connection

aelghozi, 2021-12-01:
Hello,

I am trying to make the BLE Arduino library work. I tried with all versions of ESP32 packages but usually I'm using the 1.0.6. I'm using the BLE client example from the library. Here is the serial monitor output:
`10:24:17.949 -> BLE Advertised Device found: Name: VR BOX, Address: ff:ff:20:01:b5:1e, serviceUUID: 00001812-0000-1000-8000-00805f9b34fb`
`10:24:17.949 -> Found VRBOX Server`
`10:24:17.949 -> Server has HID service`
`10:24:17.949 -> Forming a connection to ff:ff:20:01:b5:1e`
`10:24:17.949 ->  - Created client`
`10:24:18.041 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)`
`10:24:18.041 ->  - Connected to server`

The code seems to be stuck at this line: `BLERemoteService* pRemoteService = pClient->getService(serviceUUID);`

Then nothing... Sometimes it manages to connect but usually no. I also tried to modify the FreeRTOS.cpp file by using non infinite semaphores. I tried to replace `xSemaphoreTake(m_semaphore, portMAX_DELAY);` by `xSemaphoreTake(m_semaphore, 15000UL);`. Now it is not getting stuck at line `BLERemoteService* pRemoteService = pClient->getService(serviceUUID);` but the connection establishes one time each 10 times.

What can I do please ?
Any help or idea will be appreciated !

chegewara, 2021-12-01:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.h#L36-L37
Use this function instead:
`connect(BLEAdvertisedDevice* device);`
or this parameter `esp_ble_addr_type_t type = BLE_ADDR_TYPE_PUBLIC` should be `BLE_ADDR_TYPE_RANDOM` or something like that.

aelghozi, 2021-12-01:
Thank you for the reply !

I didn't catch your point because the example of the BLE ESP32 Arduino library already shows `pClient->connect(myDevice);` in the `bool connectToServer()` function.

aelghozi, 2021-12-01:
Which version of the ESP32 package do you recommend ?

chegewara, 2021-12-02:
In that case its hard to say what may be the problem. Maybe it is problem with your code or maybe library.

aelghozi, 2021-12-02:
> In that case its hard to say what may be the problem. Maybe it is problem with your code or maybe library.

My code ? It is not my code, it is the example (BLE_client) of the ESP32 BLE library **without any modification**. I just put the correct UUID here:
`// The remote service we wish to connect to.`
`static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");`
`// The characteristic of the remote service we are interested in.`
`static BLEUUID    charUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8");`

chegewara, 2021-12-03:
If it is example, then how would you like to find service with random UUID (which is in example) in your device, which is HID?
It is completely different set of services and characteristics UUID. I cant help you with that,

closed 2021-12-03 by chegewara

aelghozi, 2021-12-03:
It is not random, it is the UUIDs of my controller

khalidnk83, 2022-07-30:
I had many issues using ESP32 BLE, then I switched to nimble ble....
SO far it is very good experience, try using it from the link: https://github.com/h2zero/NimBLE-Arduino.git