nkolban/esp32-snippets #1110: Initial notification lost after reconnecting

CheapB, 2022-01-27:
I have a scenario where an ESP32 is the client and a commercial TV BLE remote is the server. The setup worked fairly well, except the battery of the remote drains too fast - about 20x faster compared with when the remote is used with a TV. 

To offset this issue, I have added code, so the client is disconnecting after allowing the remote control to go into low power mode and the client goes back in scan mode. This solves the battery issue, but when a button is pressed on the remote and the server reconnects, value of the initial key press is lost. subsequent key presses are fine. 

Any suggestions to resolve this or alternatively other ways to preserve power? The ESP32 is plugged in so I have no concerns about client power consumption.
   
` if (lastCommandValue &&((millis() - lastCommandTime) >=IDLETIMEOUT || millis() < lastCommandTime)) {
      Serial.println("Disconnecting from Remote to preserve battery life");
      BLEClient*  killClient  = BLEDevice::getClientByGattIf(lastConnId);
      Serial.println("Kill client");
      killClient->disconnect();
      doScan 		= true;
      connected 	= false;
      lastCommandValue 	= 0;
    } 
  `

chegewara, 2022-01-28:
Maybe after connecting just try to read value if its possible?
Or maybe the first key pressed is send in advertising packet?

CheapB, 2022-01-28:
Thanks for the fast response. I have tried       Serial.println(advertisedDevice.getServiceData().c_str()); in MyAdvertisedDeviceCallbacks and     Serial.println(pRemoteCharacteristic->readValue().c_str());
 in connectToServer and none of them return data. Is that what you had in mind?

chegewara, 2022-01-28:
Well, maybe not `readValue`, but yes, something like that. Right after connecting to it `getData` (or something similar function, dont remember now).