nkolban/esp32-snippets #1115: Change MAC after initialization

sovus, 2022-02-15:
Hi,

I am using the ESP32 BLE Arduino -> BLE_IBeacon example in Arduino IDE and would like to be able to change the MAC-Address of the advertising device after I initialized the BLEDevice.

Right now it looks like this:
```
uint8_t new_mac[8] = {0x01, 0x02, 0x03, 0x04, 0x05, 0x06};
esp_base_mac_addr_set(new_mac);
BLEDevice::init("ESP32");
BLEDevice::startAdvertising();
```
This changes the BLE-MAC-Address before initializing the BLEDevice. After the initialization 

`esp_base_mac_addr_set(new_mac);`

does not work anymore and the advertised mac stays the same. The only way I found to change it was to use
`ESP.restart();`
but this erases everything.

I know it is possible, because
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html#custom-base-mac
says "It is also possible to call the function esp_netif_set_mac() to set the specific MAC used by a network interface, after network initialization.", but I can't make it work.

I tried including the necessary libraries and using esp_netif_set_mac(), but I don't know how to get the esp_netif Parameter.

Can someone help or show me a working example please?

chegewara, 2022-03-21:
> to set the specific MAC used by a network interface

BLE is not considered as network interface.

You can to do it by setting private addressing mode. Im not sure if it is implemented, and if it is i just dont remember which function does it.

sovus, 2022-05-03:
> @sovus did you find a solution for this problem?

I am still looking for a soulution.

chegewara, 2022-05-05:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEAdvertising.cpp#L278