nkolban/esp32-snippets #1120: About SIG certification

rf-man, 2022-03-30:
I'm currently developing software for the ESP32-WROOM-32E,
and investigating appropriate QDID combinations to proceed with SIG certification as an End Product.
This time, I'm using "ESP32 BLE Arduino" as the software development environment, but what should I register as the QDID?
QDID Examples：
・164224：ESP-IDF Bluedroid Host
・141499：ESP NimBLE Host

I have made a similar inquiry to the Arduino side, but they have refused to respond because "ESP32 BLE Arduino" is an unofficial library.

If you have any examples of past developments, please respond.

chegewara, 2022-03-30:
This library is based on esp-idf bludroid, but if espressif refused to respond (smartass) then here you wont get more help.

rf-man, 2022-03-30:
Dear chegewara,

Thanks for the quick response.
I understand this library is based on esp-idf bludroid.

With all due respect, just to be sure,
Am I correct in assuming that you are involved in some way in the development of this library?
If you mean the opinions of those involved in the development of the library, this would be very valuable evidence in this survey.

chegewara, 2022-03-30:
Yes, you may say i was involved.

rf-man, 2022-03-30:
@chegewara 

Thank you for your response.
I will proceed to check with Espressif again based on the information you provided.

closed 2022-03-30 by rf-man

squonk11, 2022-04-02:
I am not expert on this but I think the following is correct: 
SIG certification is quite expensive: $8000,- per project. You **can not** include in one project two different SW stacks; e.g. ESP-IDF Bluedroid Host **_and_** ESM NimBLE Host; you have to decide which one you want to use. Otherwise you have to pay 2x$8000,-.  You also have to take care that you select the right QDID which fits to your ESP-IDF version. If you are using ESP32 BLE Arduino you need to check which ESP-IDF version is used for the ESP32 BLE Arduino version you are using.

rf-man, 2022-04-04:
@squonk11 
Thank you for taking the time to contact us regarding the SIG certification.
I would be careful about software version consistency.