nkolban/esp32-snippets #1122: pCharacteristic->notify() notifications are sent when value doesn't change

playrobotics-alex, 2022-04-20:
Hi!
I am communicating with a phone and once I execute the following code (only once):
`pCharacteristic->setValue(value);
          pCharacteristic->notify();`
notifications never stop coming in , although there is no change in data or anything...
Is there any way to prevent this or maybe stop notifying ?

chegewara, 2022-04-20:
Stop sending notifications.
Its your responsibility to set new value and to send notification whenever you want.

playrobotics-alex, 2022-04-20:
I send the notification only once, the problem is ios keep sending it over and over again even though a new value was not set.
Now I finally understand why this happened, so will share in case it can be useful to anyone:
inside my onNotify function I had BLE read call.
iOS doesn't provide a way to distinguish whether the characteristic has updated due to a notification or due to a read value
So it was resending the notification every time I made the read call, this created a loop because every time I got a notification I did a read call....

chegewara, 2022-04-20:
Again, its problem either with your esp32 or iOS app.
When you are sending notify you are also sending a value, not just notifying the value changed.

playrobotics-alex, 2022-04-20:
I am just trying to say the making a ble read call will make IOS send a notification every time a read call is made
Android will send a notification only when the value was changed but IOS will also send a notification every time the value is read.