nkolban/esp32-snippets #1133: Loss of bond/whitelist on reset

Juicy-AUS, 2022-06-18:
Hello,
I've set up my esp32 as a ble server using arduino ide.
I'm using code from staticpin.ino to provide some security for the server.
`  BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);` at init and
`BLESecurity *pSecurity = new BLESecurity();
  pSecurity->setStaticPIN(123456); ` after `  pServer->getAdvertising()->start();`

All works well except everytime i reset the esp32 oor power off/on and try to connect again, from what i can tell, it loses it's bonding/whitelist info.
It remains in the  "paired devices" list on the client (samsung tablet) and shows in the bonded list on nrf connect until it is powered back up, where it immediately becomes unbonded.
I assume that during the initial connection process, the esp32 tells the tablet that it's not in the whitelist so the tablet removes the bond.
I've searched a bit but haven't found a suitable fix for this. 
Any help would be appreciated.

Juicy-AUS, 2022-06-18:
I did read [https://github.com/nkolban/esp32-snippets/issues/414#issuecomment-368281667](url) but not sure where to add the ESP_LE_AUTH_REQ_SC_BOND or ESP_LE_AUTH_BOND in the `pService->start(): `.

chegewara, 2022-06-21:
https://github.com/nkolban/esp32-snippets/commit/67671174daa136c2e7cdb6dc5c316ec1b5e79495#diff-7df6f288c72f84776704a25e09e5ffcac7096cbbda9532dc646b4662700f3c4b

Juicy-AUS, 2022-07-03:
thank you. adding
`pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);` fixed issue.

closed 2022-07-03 by Juicy-AUS