nkolban/esp32-snippets #1136: BLE Library and HTTP Library Interference

HHavendano, 2022-08-01:
Hi, i've a proyect with both library. i've a code that connects to bluetooth and connects to Wi-Fi. Both work ,but not the way that i want. The codes for each library are in "setup". First i put the Bluetooth's code then the HTTPS, the problem of this is that the Bluetooth works ,but the HTTPS doesn't (it can't download a file from a website). And if i put the HTTPS over the BLE's code. HTTPS works, but BLE doesn't (I try to send a massage to ESP32 ,but it doesn't receive). 

What could be the problem? ,and what can i do?

chegewara, 2022-08-01:
memory shortage?

HHavendano, 2022-08-01:
I don't think so, i was investigating in these hours. And i found a comment that "the Bluetooth and Wifi use the same antenna or something like that" ,and that it's difficult to use it at the same time. @chegewara Do you have any idea?

chegewara, 2022-08-01:
Yes, my idea is memory shortage, because i never had any issue to use wifi and ble at the same time, and i did many such projects, including the one i am working with now. I just had to switch to NimBLE, because of memory shortage this time, and only this time, but this is huge project, including LTE.

HHavendano, 2022-08-01:
NimBLE? , My proyect is huge too.

chegewara, 2022-08-01:
https://github.com/h2zero/NimBLE-Arduino

HHavendano, 2022-08-02:
@chegewara Thanks, I'll try change the libraries.

HHavendano, 2022-08-03:
@chegewara I have problems to adapt it. i use 
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>