nkolban/esp32-snippets #1137: BLEAdvertising issues -- advertisement data is in the wrong place

bakano98, 2022-08-11:
Hi all, hoping I can get some help here as I've ran into a very perplexing issue.

The first issue is that when using `BLEAdvertising::setAdvertisementData` and `BLEAdvertising::setScanResponseData`, this issue somehow occurs:  
[image]

The next thing is that, when I use `BLEAdvertising::setManufacturerData`, the following occurs:  
[image]

I am not sure what is happening -- it would be great if someone could tell me what I am doing wrongly.

chegewara, 2022-08-11:
Yes, it looks really odd. Packets look to be swapped, but code looks good to me:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp#L151-L178
Did you check how it looks like when you scan for device?
Maybe its something wrong with nRF connect when copy device?

bakano98, 2022-08-11:
Hi @chegewara,

> Did you check how it looks like when you scan for device?
Yes. In fact, it actually looks exactly how I would want it to look like -- except for the fact that `Manufacturer Data` is in the `Advertising data` section instead of the `Scan Response data` section.

> Maybe its something wrong with nRF connect when copy device?
Unlikely -- essentially, I am trying to spoof a device by copying its advertising packet using the ESP32.  
I have tested cloning with the actual BLE device and the cloning works perfect.   
I am not sure if there is a better way to do the above as I am very new to this entire thing.

It is a necessity for the manufacturer data to be under the `Scan Response data` section -- otherwise it would not be recognised by the BLE device's app.

bakano98, 2022-08-11:
> In that case my advice is to swap scan response with advertising data on esp32.
I cant explain why it is broken like that, maybe its issue with esp-idf.

Unfortunately that does not work either 😢 

Even after swapping, the advertising packet still looks the same as in the second screenshot.

I will leave this issue open for now in case anybody has encountered anything similar -- meanwhile, I will try other methods to see if it can be resolved

bakano98, 2022-08-17:
I think there might be some underlying implementation issues with the scan response stuff.

Using [NimBLE](https://github.com/h2zero/NimBLE-Arduino) works perfectly and I am able to get the result that I want.

closed 2022-08-17 by bakano98