nkolban/esp32-snippets #1139: How to use BLE joystick to control Esp32?

Dario-Ciceri, 2022-08-22:
Hi everyone, I'm trying to use this BLE joystick (https://www.amazon.it/gp/product/B095JZ3L5J/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) as a Server to send data to Esp32 as a client. 

Now... First of all I tried this code (https://www.instructables.com/ESP32-Bluetooth-BLE-Remote-Control/) which were made for a joystick almost identical to mine BUT I have to use Esp32 core v1.0.4 or everything crashes and even if it doesn't crashes with core v1.0.4 it only connects, find the service, find the characteristics but it seems like it never gets notified even if I press buttons on my joystick.. so it doesn't work.

So I tried with Esp32 core v2.0.0, I opened BLE_Client.ino example, changed UUID's and it connects, find what it have to find but still doesn't get notified by BLE joystick... I tried my joystick on Laptop and Smartphone and everything works as supposed.

Is there something I'm missing? Do I have to ask the server to send data? if yes, how? 

Thank you in advance, I hope you could help me!

mpoelzl, 2024-12-13:
@Dario-Ciceri Did you ever find a solution for this?

Dario-Ciceri, 2024-12-13:
> @Dario-Ciceri Did you ever find a solution for this?

Yes! Initially I created a custom code although I remember with great difficulty... then I fortunately found a custom esp32 board package designed specifically to connect virtually any input device via bluetooth, I did several tests and contributed to the project, I can say without any problems that it is the best solution so far.

https://github.com/Dario-Ciceri/M5Stack-Esp32-Xbox-One-Gamepad-Demo
https://github.com/Dario-Ciceri/esp-idf-arduino-bluepad32-template
https://bluepad32.readthedocs.io/en/latest/