nkolban/esp32-snippets #1144: Data Loss occurs while sending data from ESP 32 to Mobile app using OnRead()

kpk31081998, 2022-11-05:
Hi dear niel  I am trying to send data from ESP 32 to mobile app .here I am using Light Blue android application to read my data from ESP32. I am sending my data in json format by using ArduinoJson librarie and data size I need to send is 1KB. I have used This data transfer code inside of an onRead() call back .This data which I am sending from ESP and receiving at app is not stable some times I can get complete data some times half randomly. 
But In other hand this data transfer of 1KB data from my Mobile App(Light Blue) To my ESP32 using OnWrite() callback is working very well .

Hope I will get support 
                       Thank You

chegewara, 2022-11-07:
I am almost sure its problem with size, because internal characteristics size is 600 bytes (more or less).
When you write packets are processed on the fly, but you cant read more than that.

kpk31081998, 2022-12-06:
Thank you so much for your replay dear chegewara Is there any way to Send my 1KB data using my ESP32 BLE to  android app

chegewara, 2022-12-06:
Not in one packet/transmission, you have to split data.
BTW recently i found that nimble characteristics data size is up to 512 bytes. Also someone reported that its the size on android these days

kpk31081998, 2022-12-06:
Okay  please don't mind I am not much familiar with BLE  can you give me a idea to do this splitting.  can I write two set statements in onRead() method of my characteristic callback? or is there any other way? .I am sending data in JSON format using ArduinoJson librarie is it can do any help?

kpk31081998, 2022-12-08:
Thanks for you support filled with  patience but I would like to admit that I know nothing about android  But I using Light Blue to test my ESP32 .  

class MyCallbacks2: public BLECharacteristicCallbacks {
 void onRead(BLECharacteristic *pCharacteristic2){
    preferences.begin("alarms", false);
         // do stuff here on connecting
       StaticJsonDocument<384> doc;
      preferences.end();
      serializeJson(doc, Eventlog);
     pCharacteristic2->setValue(Eventlog.c_str());
}
};  
Here above is the code snippet I am using to send data to app
Here after making it into two JSON strings am I can call this setValue() method two times?

chegewara, 2022-12-08:
> Here after making it into two JSON strings am I can call this setValue() method two times?

No, because second time will only overwrite first set value.

Idea:
- what if you would track how many times value is read from characteristic?
- then on odd read you setValue with some data and on even read you setValue with different data?
- you can split is 2, 3, 4 times, but the idea is similar

kpk31081998, 2023-10-09:
thank for support @chegewara

closed 2023-10-09 by kpk31081998