nkolban/esp32-snippets #1147: start service to be visible after an event.. help (or stop)

Sladerix, 2023-02-27:
Hi every one, I need to recreate this behavior:

1. The BLE Server have 2 services:
    **ServiceA** has 1 characteristic called CODE,
    **ServiceB** has _n_ characteristics

2. ESP32 boots up and starts only **ServiceA**

3. Now, a client can write on the CODE characteristic, if the code is correct **ServiceB** should start and appear in the available services..

I want to make something like this to create mini-security system... I already tried to do that but it does not work... in the moment I call `pServiceB->start(); ` the client (in this case smartphone with nRF Connect App) start loading until disconnection, and ESP32 freezes.

this is the code:
```cpp
#define SERVICE_UUID                   "B04EFAC2-E040-4AC1-A464-393508E6C2C9"
#define LEDSERVICE_UUID                "9C5D8FB7-9D89-4EFC-BBCB-D1A98E267354"

#define LED_1_UUID                     "EDF849DE-70C0-4A79-8F36-A199AEB3B2AA"
#define LED_2_UUID                     "43081C63-021C-4311-A03C-AC91327BD808"
#define LED_3_UUID                     "50E4902F-E29D-4193-AD7B-DC3E3ACBD0CE"
#define ACCESS_CODE_UUID               "E37ADC7D-4E0F-466E-8674-02FD0F8983BB"

BLEServer *pServer;
BLEService *pService;
BLEService *pLedService;
BLECharacteristic *led1Characteristic;
BLECharacteristic *led2Characteristic;
BLECharacteristic *led3Characteristic;
BLECharacteristic *accesscodeCharacteristic;

void setup() {
  BLEDevice::init("ESP32 Server");
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  pService = pServer->createService(BLEUUID(SERVICE_UUID), 3, 0);
  accesscodeCharacteristic = pService->createCharacteristic(ACCESS_CODE_UUID, BLECharacteristic::PROPERTY_WRITE);
  accesscodeCharacteristic->setCallbacks(new accessCodeCallbacks());
  pService->start();

  pLedService = pServer->createService(BLEUUID(LEDSERVICE_UUID), 17, 0);
  led1Characteristic = pLedService->createCharacteristic(
        LED_1_UUID,
        BLECharacteristic::PROPERTY_WRITE |
        BLECharacteristic::PROPERTY_READ);

  led2Characteristic = pLedService->createCharacteristic(
        LED_2_UUID,
        BLECharacteristic::PROPERTY_WRITE |
        BLECharacteristic::PROPERTY_READ);
                                                    
  led3Characteristic = pLedService->createCharacteristic(
        LED_3_UUID,
        BLECharacteristic::PROPERTY_WRITE |
        BLECharacteristic::PROPERTY_READ);

  led1Characteristic->setCallbacks(new ledCallbacks());
  led2Characteristic->setCallbacks(new ledCallbacks());
  led3Characteristic->setCallbacks(new ledCallbacks());
  // DO NOT START THIS pLedService!! Later.

  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  BLEAdvertisementData pAdvertisementData;
  pAdvertisementData.setManufacturerData(getBluetoothAddress());
  pAdvertisementData.setAppearance(ESP_BLE_APPEARANCE_GENERIC_COMPUTER);

  pAdvertising->setAdvertisementData(pAdvertisementData);
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->addServiceUUID(LEDSERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);
  pAdvertising->setMinPreferred(0x12);
  
  BLEDevice::startAdvertising();
}

void loop() {}

void event() {
/* THIS FUNCTION REPRESENT AN EVENT, e.g. code wrote in the access code characteristic that I will don't write to not be too long */

pLedService->start();
// HERE THE PROBLEM HAPPEN, SERVICE DOES NOT BECOME VISIBILE AND EVERYTHING FREEZES...
}
```

Can someone help me figure why it doesn't work, and/or how to get that behavior working correctly? thanks in advance

chegewara, 2023-02-28:
Did you try to start this service not from BLE event or callback, just from `loop()` or other task?

Sladerix, 2023-02-28:
> Did you try to start this service not from BLE event or callback, just from `loop()` or other task?
I actually found a workaround, it seems to work outside the ble-related events. 
later I will upload a working sample code

chegewara, 2023-02-28:
This is what i am suggesting.

closed 2023-02-28 by chegewara

Sladerix, 2023-02-28:
> This is what i am suggesting.

When can I find documentation about what i **can** and **cannot** do in those functions ?

chegewara, 2023-02-28:
I dont think espressif is providing such docs, you just have to figure it our.