nkolban/esp32-snippets #1152: GATT table size checking / setting

ravinagaraj61, 2023-03-04:
I have an ESP32 BLE server and a BLE32 BLE client talking to each other. The server, which uses the Arduino ESP32 BLE library, has five Characteristics, three Read and notify and two write. The Client is handling them correctly. When I introduce a 2904 descriptor into any one of the Read, Notify Characteristics, the Client stops being able to talk to the last Characteristic, whichever that is in the creation process (at end of GATT, I assume). The Client's code is not changed. The Server and Client are not bonded. I have changed the order in which the Characteristics are declared and found that the last is always affected. My question is: In this Arduino ESP32 BLE library, is there a function I can use to check GATT size / usage or to allocate more memory to GATT? There is a large amount of unused memory available in the ESP32. 2902 descriptors don't seem to cause a problem. I have verified the client behavior with the nRF Android App as well.

chegewara, 2023-03-04:
Actually i cant say how many handles currently is available, but it would be easier if you tell us how many characteristics and descriptors are you using in overall.

ravinagaraj61, 2023-03-04:
Here is the code for registering the Characteristics in the GATT. Note that last Characteristic (humidity) has a 2904 descriptor defined but the Temperature characteristic has it commented out. This code works but there is no 2904 descriptor visible for Humidity in the nRF app. If I uncomment the 2904 descriptor for temperature, the Humidity characteristic is no longer visible in the nRF app. If I put the write Characteristics last, one of them drops off the edge.

----
  BLEService *pService = pServer->createService(SERVICE_ENVIRONMENTAL_SENSING_UUID);
  // Create BLE Characteristics
  
  pSetLoThreshCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID_LoThresh,
                      BLECharacteristic::PROPERTY_WRITE
                    );

  pSetLoThreshCharacteristic->setCallbacks(new MyLoThreshCallback());

                      
  pSetHiThreshCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID_HiThresh,
                      BLECharacteristic::PROPERTY_WRITE
                    );

  pSetHiThreshCharacteristic->setCallbacks(new MyHiThreshCallback());

  pStatCharacteristic = pService->createCharacteristic(
                     CHARACTERISTIC_UUID_STATUS,
                     BLECharacteristic::PROPERTY_READ   |
                     BLECharacteristic::PROPERTY_NOTIFY
                    );
  pStatCharacteristic->addDescriptor(new BLE2902());

  BLE2904 * pStatDescriptor = new BLE2904();
  pStatCharacteristic->addDescriptor(pStatDescriptor);
  pStatDescriptor->setFormat(BLE2904::FORMAT_UINT8);
  pStatDescriptor->setUnit(0x2700); //Unitless

  ptempCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_TEMPERATURE_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_NOTIFY 
                    );
                   
  ptempCharacteristic->addDescriptor(new BLE2902());
/*
  BLE2904 * ptempDescriptor = new BLE2904();
  ptempCharacteristic->addDescriptor(ptempDescriptor);
  ptempDescriptor->setFormat(BLE2904::FORMAT_SINT16);
  ptempDescriptor->setExponent(-2); //Divide this by 100
  ptempDescriptor->setUnit(0x272F); //Temperature in Celsius per BLE SIG
*/                
  phumidCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_HUMIDITY_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_NOTIFY 
                    );

  phumidCharacteristic->addDescriptor(new BLE2902());

  BLE2904 * phumidDescriptor = new BLE2904();
  phumidCharacteristic->addDescriptor(phumidDescriptor);   
  phumidDescriptor->setFormat(BLE2904::FORMAT_UINT16);
  phumidDescriptor->setExponent(-2); //Divide this by 100
  phumidDescriptor->setUnit(0x27AD); //Percentage per BLE SIG

ravinagaraj61, 2023-03-04:
The environment is an ESP32-C3-DevkitM-1 V1.0 from Espressif

chegewara, 2023-03-05:
I remember now. This function has additional argument with default value:
```
pServer->createService(SERVICE_ENVIRONMENTAL_SENSING_UUID, handles);
```
You should increase `handles` value.

ravinagaraj61, 2023-03-06:
Ah, thanks a lot, I will try that.

closed 2023-03-06 by ravinagaraj61