nkolban/esp32-snippets #1155: Issue with re initializing BLE device in esp32

DharmikEC, 2023-03-23:
I am using ESP32 and arduino 2.0.4 IDE. My goal is to stop BLE for http request and than re initiate BLE again. I am performing these both in single task separate from loop.
I have gone through several posts on this topic but could not find solution. Also those posts are quiet old.

From what I have gathered, 
I can not call `BLEDevice::deinit(true);`  because it will erase memory and then `BLEDevice::init();` also wont work after this. Is this still correct ? or there is an update in library ? 
If I call `BLEDevice::deinit(false);` then it will not clear the memory and I will be able to use BLE again with `BLEDevice::init();`. However, doing this multiple time also result in error in http request. After few attempt it will reboot the esp32. From what I have read, others faced this issue too. (They used word "memory leak" which I do not understand.)
```
15:48:05.821 -> [613636][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():267]: (-32512) SSL - Memory allocation failed
15:48:05.821 -> [613637][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -32512
```
I have tried this with only BLE in code. I had loop running with enabling and disabling BLE after every one second delay. And I also get error there too, not surprisingly after lot of attempts than in task. So it is a memory issue after all.
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
```
Now again my question is, How can I re initiate BLE device after clearing memory ? Is there literally no way for doing this ?
Is there any other way to do what I am trying to do ?

DharmikEC, 2023-03-23:
> The simplest answer is `dont`. Just dont do it, because it wont help you, especially if you dont know what you are doing. Its better if you switch to nimble library. You can use the same code, just different headers and you will have more free memory.

Thank you for quick reply. I am afraid I cant drop this idea. At last, I just have to go with bluetooth classic which works perfectly fine. Just more free memory will not help as device will stay on for months not just few days and I can not limit memory there. 

If there is any other approach, then I'll be glad if you can share link or any other pointers. I am still new but I definitely like to give this a try.

chegewara, 2023-03-23:
It will be very hard to write bluetooth classic + https, especially that you probably have more features.
In that case memory will be your biggest problem, even if you know how to write good code.

`memory leak` -
https://en.wikipedia.org/wiki/Memory_leak

DharmikEC, 2023-03-23:
I have already implemented classic Bluetooth with http. I am trying do the same with BLE for more functionality. Loosing memory on every single switch is what I can not deal with. So I am hoping there is solution to this problem as well. 

I don't understand, what happens when memory is allocated for BLE each time ESP restarts that I can not replicate the same in the code while it is running.