nkolban/esp32-snippets #1168: BLEwrite Example -> onWrite not working

JanB97, 2023-07-02:
So I've been banging my head against a wall and don't understand the problem here.
If been trying the Write and the UART example but on both the onWrite even is not fired after writing something to the characteristic via the nRF_Connect App.
onRead does work on both.

Can someone guide me what I need to change in the examples to make it work?
I am using a Lolin D32.

JanB97, 2023-07-02:
This is what I get from the Verbose Debugging on the ESP..
It does recognize if I send something, but the funktion never get's executed.

21:31:12.254 -> [D][BLEDevice.cpp:104] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
21:31:12.287 -> [V][BLEUtils.cpp:1531] dumpGattServerEvent(): GATT ServerEvent: Unknown
21:31:12.287 -> [V][BLEUtils.cpp:1721] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
21:31:12.287 -> [D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
21:31:12.287 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
21:31:12.287 -> [D][BLEValue.cpp:46] addPart(): >> addPart: length=1
21:31:12.321 -> [D][BLECharacteristic.cpp:292] handleGATTServerEvent():  - Response to write event: New value: handle: 2a, uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8
21:31:12.321 -> [D][BLECharacteristic.cpp:295] handleGATTServerEvent():  - Data: length: 1, data: 25
21:31:12.321 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
21:31:12.321 -> [D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
21:31:12.354 -> [D][BLEDevice.cpp:104] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
21:31:12.354 -> [V][BLEUtils.cpp:1531] dumpGattServerEvent(): GATT ServerEvent: Unknown
21:31:12.354 -> [V][BLEUtils.cpp:1721] dumpGattServerEvent(): dumpGattServerEvent: *** NOT CODED ***
21:31:12.354 -> [D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
21:31:12.392 -> [D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
21:31:12.392 -> [D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
21:31:12.392 -> [D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent

JanB97, 2023-07-02:
Okay so I tested it with a new Heltec Lora ESP32 V3.
With this board it works, it has a ESP32S3.
I am still interested why it is not working on the Lolin D32 Board.

LukaszWiktor, 2023-12-30:
I'm struggling with the same issue. The `onWrite` callback doesn't work on my Seeed Xiao ESP32-C3 board, but it's working fine on Seeed Xiao ESP32-S3.

evanderkuy, 2024-03-05:
I am having a similar issue using a Waveshare ESP32-S3 mini. When I run the BLEwrite Example the onWrite event is never triggered on Platformio (Ubuntu). But when I run it on Arduino IDE (Windows) with the same code it is working fine.

As the Waveshare board is not on the platform I am using lolin_s3_mini with below ini file.

[env:lolin_s3_mini]
; platform = espressif32
platform = espressif32@6.5.0
board = lolin_s3_mini
board_upload.flash_size = 4MB
build_flags = 
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DARDUINO_USB_MODE=0
	-DARDUINO_RUNNING_CORE=1
framework = arduino
lib_deps = 

upload_protocol = esp-prog
upload_port = //dev/ttyUSB2
upload_speed = 921600
debug_tool = esp-prog
monitor_port = //dev/ttyUSB0
monitor_speed = 115200