nkolban/esp32-snippets #1170: SPI CLOCK error between ST7920 <-> ESP32 WROOM32

willianaraujo, 2023-07-22:
I'm using an ESP32 WROOM32 DEVKIT board, a 128x64 display with ST7920 chip. My development environment is ESP-IDF in VSCode.
I created a project from scratch, added only the U8G2 library directly from the official repository on GitHub following the official documentation. I also added Kolban's component, the u8g2_esp32_hal to interface with the main u8g2 library.

The first time I used this combination of libraries, still last year, it worked perfectly.... after approximately 6 months without running the development environment, I returned to work on the project, reinstalled VSCode, ESP-IDF, U8G2, and U8G2_esp32_hal, all from the official repositories in their current versions. However, now my CPU started to restart repeatedly when I try to start SPI communication between the ESP32 and the ST7920.
I even suspected the hardware, so I got a new DEVKIT board, a new display, and connected the display directly to the board, without any peripherals, just display-esp, and the board keeps resetting...

My boot log is as follows:
```markdown
0x40080400: _init at ??:?

load:0x40080404,len:3876
entry 0x4008064c
.I (32) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (33) boot: compile time Jul 21 2023 15:13:26
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b9d0h ( 47568) map
I (124) esp_image: segment 1: paddr=0001b9f8 vaddr=3ffb0000 size=0226ch (  8812) load
I (127) esp_image: segment 2: paddr=0001dc6c vaddr=40080000 size=023ach (  9132) load
I (133) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=17db0h ( 97712) map
I (173) esp_image: segment 4: paddr=00037dd8 vaddr=400823ac size=0b2c8h ( 45768) load
I (199) boot: Loaded app from partition at offset 0x10000
I (199) boot: Disabling RNG early entropy source...
I (211) cpu_start: Multicore app
I (211) cpu_start: Pro cpu up.
I (211) cpu_start: Starting app cpu, entry point is 0x4008119c
0x4008119c: call_start_cpu1 at /home/wka/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (198) cpu_start: App cpu up.
I (229) cpu_start: Pro cpu start user code
I (229) cpu_start: cpu freq: 160000000 Hz
I (229) cpu_start: Application information:
I (234) cpu_start: Project name:     ST7920
I (239) cpu_start: App version:      1
I (243) cpu_start: Compile time:     Jul 21 2023 15:12:36
I (249) cpu_start: ELF file SHA256:  71d26c1d383068f3...
I (255) cpu_start: ESP-IDF:          v5.1-dirty
I (261) cpu_start: Min chip rev:     v0.0
I (265) cpu_start: Max chip rev:     v3.99 
I (270) cpu_start: Chip rev:         v3.1
I (275) heap_init: Initializing. RAM available for dynamic allocation:
I (282) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (288) heap_init: At 3FFB2F18 len 0002D0E8 (180 KiB): DRAM
I (294) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (301) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (307) heap_init: At 4008D674 len 0001298C (74 KiB): IRAM
I (315) spi_flash: detected chip: generic
I (318) spi_flash: flash io: dio
W (322) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (336) app_start: Starting scheduler on CPU0
I (340) app_start: Starting scheduler on CPU1
I (340) main_task: Started on CPU0
I (350) main_task: Calling app_main()
I (350) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
I (360) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
E (370) esp_clk_tree: esp_clk_tree_src_get_freq_hz(21): unknown clk src
E (370) spi_master: spi_bus_add_device(368): invalid sclk speed
E (380) err: esp_err_t = 258

assert failed: u8g2_esp32_spi_byte_cb u8g2_esp32_hal.c:89 (0 && "spi_bus_add_device(HOST, &dev_config, &handle_spi)")
[... 165 lines cut ...]
      // ESP_LOGI(TAG, "... Adding device bus.");
      ESP_ERROR_CHECK(spi_bus_add_device(HOST, &dev_config, &handle_spi));

      break;
    }

    case U8X8_MSG_BYTE_SEND: {
      spi_transaction_t trans_desc;
      trans_desc.addr = 0;
      trans_desc.cmd = 0;
      trans_desc.flags = 0;
      trans_desc.length = 8 * arg_int;  // Number of bits NOT number of bytes.
      trans_desc.rxlength = 0;
      trans_desc.tx_buffer = arg_ptr;
      trans_desc.rx_buffer = NULL;

      // ESP_LOGI(TAG, "... Transmitting %d bytes.", arg_int);
      ESP_ERROR_CHECK(spi_device_transmit(handle_spi, &trans_desc));
      break;
    }
  }
  return 0;
}  // u8g2_esp32_spi_byte_cb
```
Why can't I even run the example codes anymore, whereas last year everything worked perfectly, I even developed several applications with screens, menus, and animations on this same hardware...

willianaraujo, 2023-07-22:
I kept looking for the issue and found the root cause, but I'm not exactly sure why the problem started happening. I believe it could be due to an update in ESP-IDF or FreeRTOS, someone with more experience in embedded systems might know...

Basically, in the u8g2_esp32_spi_byte_cb() function when filling out the fields of the dev_config variable to send to the spi_bus_add_device() function, none of the documentation I found instructed to pass the dev_config.clock_source field, but from what I saw in the functions that are called from here, this information is needed.

So, the error above was caused because the dev_config.clock_source field is not being filled in the u8g2_esp32_spi_byte_cb function before passing the dev_config as a parameter in the line ESP_ERROR_CHECK(spi_bus_add_device(HSPI_HOST, &dev_config, &handle_spi));

This way the initialization for the dev_config parameter would look like this:

```C++

		  spi_device_interface_config_t dev_config;
		  dev_config.address_bits     = 0;
		  dev_config.command_bits     = 0;
		  dev_config.dummy_bits       = 0;
		  dev_config.mode             = 0;
		  dev_config.duty_cycle_pos   = 0;
		  dev_config.cs_ena_posttrans = 0;
		  dev_config.cs_ena_pretrans  = 0;
		  dev_config.clock_speed_hz   = 10000;
		  dev_config.clock_source     = SPI_CLK_SRC_DEFAULT; //SOC_MOD_CLK_APB;
		  dev_config.spics_io_num     = u8g2_esp32_hal.cs;
		  dev_config.flags            = 0;
		  dev_config.queue_size       = 200;
		  dev_config.pre_cb           = NULL;
		  dev_config.post_cb          = NULL;
		  //ESP_LOGI(TAG, "... Adding device bus.");
		  ESP_ERROR_CHECK(spi_bus_add_device(HSPI_HOST, &dev_config, &handle_spi));
```

That's it. Looking at the current ESP-IDF code, I can't understand how my previous code worked without this field, as browsing through previous versions of ESP-IDF on GitHub, this information was always mandatory.

I won't mark this thread as resolved since, as I mentioned, I don't understand why it worked before. Therefore, my solution might not be the correct one, or everything I said may not make sense, but as I'm behind schedule with my project, I won't delve further into this investigation.

Cheers!

William-Irvine, 2023-08-21:
I have noticed this issue as well.

Using Kolban's HAL I get the same crash after updating from 4.2 to 5.1, but only when I have logging set to INFO or higher. 

Otherwise I get the same crash.

Going to try the your solution now.

William-Irvine, 2023-08-21:
I added the clock source and it built with logging set to none.

Was still flaky telling me the argument was invalid, so I put a 2 second delay before my call to "u8g2_Setup_st7920_128x64_f();" and now it seems to run perfectly.

I am wondering if it has to do with how fast the device is booting now, and thinking maybe its trying to set that clock source before the chip can initialize it?

Either way its interesting. 

Thanks for the initial post, helped me a lot.

fherrera124, 2024-06-30:
the bug is in the line:

`spi_device_interface_config_t dev_config;`

in `u8g2_esp32_hal.c`. Members that are not set, have garbage values.

the solution is to use memset, or simply initialize the variable with:

`spi_device_interface_config_t dev_config = {0};`

that assures that all members of `dev_config` are intialized with 0; Then `spi_bus_add_device()` will detect that the member `.clock_source` is cero and set the value `SPI_CLK_SRC_DEFAULT`.

closed 2025-03-08 by willianaraujo