nkolban/esp32-snippets #1183: Is it possible to scan and detect smartphones?

Tirititelu, 2024-02-11:
Hello everyone,

I have been testing (a lot) and reading a lot about BLE scanning and I am unable to detect any smartphone, with BLEScan or NimBLEScan sketches. At some points I have come to think that this is not possible, since from what I have read client devices, like smartphones, do not emit almost never to save power, but at least it should detect them when BLE is activated on them, as happens with any laptop, which does detect smartphones when you activate BLE.

I have gone through all the posts on this forum and Neil Kolban's post and have not found the solution. I have been playing with the setPower() function, without success, it does not detect any smartphone.

So my question is simple, is it possible to detect them somehow with the ESP32 or is there any reason why it does not detect them. As I said at the beginning, if my laptop detects smartphones, I understand that ESP32 should too, but surely I'm missing something.

Thank you very much in advance.

Regards

chegewara, 2024-02-12:
Yes, if your laptop can detect smartphone then esp32 can detect it too, as long as its BLE nor classic bluetooth.
So make sure it is BLE.

For classic bluetooth you need different library, if any exist, or esp-idf API.

Tirititelu, 2024-02-12:
Thank you for your reply Chegewara.

I am testing with four different smatphones:

OPPO Find X5 Lite: BLE 5.2
MOTOROLA G32: BLE 5.2
SAMSUNG A8: BLE 5.0
BQ Aquairs X5: BLE 4.0

And with two different boards, one with an ESP32 (Lilygo T-Call SIM800L) and ESP32S3 (Heltec Wireless Stick Lite(V3)) I can't detect any of them. As I said I have used the "BLE_Scan" sketch example of the Arduino IDE (Neil Kolban), the modified NimBLE library (h2zero) and the bt_discover of the Espressif IDE.

I have also been "playing" with the setPower function, with no result.

Let's see if you can think what could be happening.

Thank you very much.

Regards

chegewara, 2024-02-12:
There is simple way to test it. 
If you know how to use nRF connect on android then you should know how to use advertiser tab.

Tirititelu, 2024-02-12:
Thanks again for your reply, chegewara.

If I understood you correctly, please correct me if I'm wrong, is that I installed on one of my smartphones the application, nRF connect to configure in the "advertiser" tab that this behaves in this way. I have done that, and then if it is detected by the BLEscanner of my ESP32, but I do not know or do not understand if that helps me at all.

What I am trying or want to know if it is possible, is if my ESP32 is able to detect my smartphones as well as my laptop or other smartphones between them, without the need to install anything additional on them. My laptop (and any other) detects my smartphones if they have the BLE enabled (without the need of nRF connect), so I understand, that the ESP32 should also, but I can't find the way.

Thanks again.

chegewara, 2024-02-12:
This is only to show you that you need 3rd party app on smartphone to advertise BLE packets.
Thats why it is LE, low energy, which means its not advertising all the time.

Alse, like i said, laptop most likely can detect classic bluetooth.

Tirititelu, 2024-02-12:
Hello again chegewara,

But that is not what you describe what is happening. My laptop, any laptop I test or any other smartphone DOES detect my smartphones when I activate the BLE in them without the need of any application like nRF connect, which is not happening with the ESP32.

Therefore, it leads me to the conclusion that ESP32 does not detect smartphones EVER (if they do not have a third party application) that with the code and libraries I am using is not enough.

Any other ideas. I will continue to investigate.

Thank you very much.

chegewara, 2024-02-12:
Sorry, but i have no idea how to explain it to you, since you dont read my words.
Laptop can detect `CLASSIC` bluetooth when you enable bluetooth in you smartphone.
This is bluetooth LE library, so is NimBLE.

If you can prove i am wrong then i am willing to learn something new.