nkolban/esp32-snippets #1194: Stack Smashing

varun2200, 2024-06-18:
13:36:40.995 -> BLE Advertised Device found: Name: , Address: 4f:94:11:6f:26:4e, manufacturer data: 4c001005229870d425, txPower: 8, rssi: -74
13:36:40.995 -> BLE Advertised Device found: Name: , Address: 06:ad:cd:18:de:2f, manufacturer data: 06000109212a5b65e3ab75c44a6967734c656e6f766f, rssi: -96
13:36:42.531 -> Found our service
13:36:42.531 ->
13:36:42.531 -> Stack smashing protect failure!
13:36:42.531 ->
13:36:42.531 ->
13:36:42.531 -> Backtrace: 0x40082399:0x3ffc84c0 0x4009372d:0x3ffc84e0 0x400823aa:0x3ffc8500 0x400d5e92:0x3ffc8520 0x400d5f91:0x3ffc85c0 0x400d68c7:0x3ffc8610 0x400d69fe:0x3ffc86b0 0x400d235d:0x3ffc8710 0x400d256d:0x3ffc8750 0x400da2e0:0x3ffc8770

the BLE client application is restarting again and again when it attempts to connect to an ios device on which the server is hosted 
And there is a stack smashing protect failure 
I am not able to find where the issue is.
If the ble library is compatible with ios devices, it should work

RASPIAUDIO, 2024-06-19:
exact same bug here

RASPIAUDIO, 2024-06-20:
downgarded to ESP version 2.0.17 and no more crash but still no notification

HexfeT78, 2024-06-22:
Find the following function in the BLERemoteCharacteristic.cpp file and replace it with it. The problem will be solved.

void BLERemoteCharacteristic::retrieveDescriptors() {
  log_v(">> retrieveDescriptors() for characteristic: %s", getUUID().toString().c_str());

  // Remove any existing descriptors.
  removeDescriptors();  

  uint16_t offset = 0;
  esp_gattc_descr_elem_t result[10];  // Array to hold multiple results.
  while (true) {
    uint16_t count = 10;  // Number of descriptors to retrieve in each call.
    esp_gatt_status_t status = ::esp_ble_gattc_get_all_descr(
      getRemoteService()->getClient()->getGattcIf(), 
      getRemoteService()->getClient()->getConnId(), 
      getHandle(), 
      result, 
      &count, 
      offset
    );

    if (status == ESP_GATT_INVALID_OFFSET) {  // End of the entries.
      break;
    }

    if (status != ESP_GATT_OK) {
      log_e("esp_ble_gattc_get_all_descr: %s", BLEUtils::gattStatusToString(status).c_str());
      break;
    }

    if (count == 0) {
      break;
    }

    for (int i = 0; i < count; i++) {
      log_d("Found a descriptor: Handle: %d, UUID: %s", result[i].handle, BLEUUID(result[i].uuid).toString().c_str());

      BLERemoteDescriptor *pNewRemoteDescriptor = new BLERemoteDescriptor(result[i].handle, BLEUUID(result[i].uuid), this);
      m_descriptorMap.insert(std::pair<String, BLERemoteDescriptor *>(pNewRemoteDescriptor->getUUID().toString(), pNewRemoteDescriptor));
    }

    offset += count;  // Increment the offset by the number of descriptors retrieved.
  }

  log_v("<< retrieveDescriptors(): Found %d descriptors.", m_descriptorMap.size());
}

RASPIAUDIO, 2024-06-24:
thanks for your answer but it did not solved the issue. 
To fix it I removed the warnings :

bool setupCharacteristics(BLERemoteService* pRemoteService, NotifyCallback pNotifyCallback)
{
  // Get all the characteristics of the service using the handle as the key
  pmapbh = pRemoteService->getCharacteristicsByHandle();
  
  // Only interested in report characteristics that have the notify capability
  for (itrbh = pmapbh->begin(); itrbh != pmapbh->end(); itrbh++)
  {
    BLEUUID x = itrbh->second->getUUID();
    Serial.print("Characteristic UUID: ");
    Serial.println(x.toString().c_str());
    
    // The UUID must match the report UUID
    if (ReportCharUUID.equals(itrbh->second->getUUID()))
    {
      // Found a report characteristic
      Serial.println("Found a report characteristic");

      if (itrbh->second->canNotify())
      {
        Serial.println("Can notify");
        // Register for notifications from this characteristic
        itrbh->second->registerForNotify(pNotifyCallback);

        sprintf(bfr, "Callback registered for: Handle: 0x%08X, %d", itrbh->first, itrbh->first);
        Serial.println(bfr);
      }
      else
      {
        Serial.println("No notification");
      }
    }
    else
    {
        sprintf(bfr, "Found Characteristic UUID: %s\n", itrbh->second->getUUID().toString().c_str()); 
        Serial.println(bfr);
    }
  }
  
  return true; // Indicate success
}

bool connectToServer()
{
  Serial.print("Forming a connection to ");
  Serial.println(myDevice->getAddress().toString().c_str());
  
  BLEClient*  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");

  pClient->setClientCallbacks(new MyClientCallback());

  // Connect to the remote BLE Server.
  pClient->connect(myDevice);  // If you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
  Serial.println(" - Connected to server");

  // BLE servers may offer several services, each with unique characteristics
  // We can identify the type of service by using the service UUID
  
  // Obtain a reference to the service we are after in the remote BLE server.
  // This will return a pointer to the remote service if it has a matching service UUID
  BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr)
  {
    Serial.print("Failed to find HID service UUID: ");
    Serial.println(serviceUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }

  Serial.println(" - Found HID service");
  setupCharacteristics(pRemoteService, notifyCallback);
  
  pRemoteService = pClient->getService(BatteryServiceUUID);
  if (pRemoteService == nullptr)
  {
    Serial.print("Failed
to find battery service UUID: ");
Serial.println(BatteryServiceUUID.toString().c_str());
}
else
{
Serial.println(" - Found battery service");
setupCharacteristics(pRemoteService, BatteryNotifyCallback);
}

connected = true;
return true; // Indicate success
}