nkolban/esp32-snippets #1196: 'ringbuf_type_t' has not been declared!

RashmikaDushan, 2024-06-21:
With the given examples this error always occurs. "'ringbuf_type_t' has not been declared"
rduino/libraries/ESP32_BLE_Arduino/src/FreeRTOS.h:61:35: error: 'ringbuf_type_t' has not been declared
   61 |         Ringbuffer(size_t length, ringbuf_type_t type = RINGBUF_TYPE_NOSPLIT);

ManKhunt, 2024-07-02:
Hello Dushan,
I'm also getting same error..... **"ringbuf_type_t"**
If you resolve this problem then help me out solve same error.

RashmikaDushan, 2024-07-02:
> Hello Dushan, I'm also getting same error..... **"ringbuf_type_t"** If you resolve this problem then help me out solve same error.

I just delete the esp32-ble-arduino library and used the one included in esp32 board(from board manager).

strgPeter, 2024-11-01:
any specific board?